dusty-nv/jetson-inference #1496: Incorporating deepsort in detectnet.py

RoninHunter, 2022-09-13:
In the below example of deepsort setup you can see it requires a "frame" to produce its tracks.

Can I use the "img" produced by input.Capture() as a frame or is there more layers of complexity to img? 

```
from deep_sort_realtime.deepsort_tracker import DeepSort
tracker = DeepSort(max_age=5)
bbs = object_detector.detect(frame) 
tracks = tracker.update_tracks(bbs, frame=frame) # bbs expected to be a list of detections, each in tuples of ( [left,top,w,h], confidence, detection_class )
for track in tracks:
    if not track.is_confirmed():
        continue
    track_id = track.track_id
    ltrb = track.to_ltrb()
```

I noticed in detectnet-snap there is a little bit of CUDA acrobatics happing there to save a cropped image. Would something similar be required to treat the img as a frame for processing in deepsort?

Also, I see for the jetson there is a "deepstream" which has deepsort already incorporated into it. Would you recommend this within the context of detectnet?

dusty-nv, 2022-09-13:
I'm not sure what format image `deep_sort_realtime` expects, whether that is a numpy array or PyTorch tensor, ect.   If it's a numpy array, then you could use `jetson.utils.cudaToNumpy()` to map the CUDA image to numpy, and then pass it to deepsort.  I don't believe any cropping/ect would be necessary.    

This does look like an interesting effort - I had previously scoped integrating a TensorRT (https://github.com/RichardoMrMu/deepsort-tensorrt, but it wasn't modular or easy to incorporate).  So if you were to get this working, I'd probably like to take another look.

DeepStream does support DeepSort, as well as other tracking algorithms, and has superior performance.

RoninHunter, 2022-09-15:
jetson.utils.cudaToNumpy() Has indeed helped. I was able to read in the image frame.

However, I am now trying to create cropped chips of each detection as seen in the following example.

```
from deep_sort_realtime.deepsort_tracker import DeepSort
tracker = DeepSort(max_age=5)
bbs = object_detector.detect(frame) # your own object detection

object_chips = chipper(frame, bbs) # your own logic to crop frame based on bbox values
embeds = embedder(object_chips) # your own embedder to take in the cropped object chips, and output feature vectors

tracks = tracker.update_tracks(bbs, embeds=embeds) # bbs expected to be a list of detections, each in tuples of ( [left,top,w,h], confidence, detection_class ), also, no need to give frame as your chips has already been embedded
for track in tracks:
    if not track.is_confirmed():
        continue
    track_id = track.track_id
    ltrb = track.to_ltrb()
```

Which CUDA cpp file can I find the implementation of cudaToNumpy and is there something there that can help me crop the image frame before handing it to an embedder?

dusty-nv, 2022-09-15:
See detectnet-snap.py for an example of cropping images of the detected objects:   https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet-snap.py

RoninHunter, 2022-09-16:
Ah yes I was just looking at that. Thank you dusty

closed 2023-04-27 by dusty-nv