dusty-nv/jetson-inference #1498: imagenet:  failed to initialize imageNet

lijoe123, 2022-09-14:
Hello, when i used the jetson xavier nx to run the command 
`./imagenet-console ./images/bird_0.jpg output.jpg`
it occur this problem:
[image]
[image]
**And this is my environment**:
[image]
Thank you!!!!

dusty-nv, 2022-09-14:
Hi @lijoe123, which specific Jetson Xavier NX and version of JetPack-L4T are you running?

Due to the `unknown embedded device detected` error message, I think you may need to upgrade your JetPack installation.

lijoe123, 2022-09-15:
> Hi @lijoe123, which specific Jetson Xavier NX and version of JetPack-L4T are you running?
> 
> Due to the `unknown embedded device detected` error message, I think you may need to upgrade your JetPack installation.

Thank you for your answer. This is my jetpack version
[image]

dusty-nv, 2022-09-15:
Which variant of Jetson Xavier NX are you running?   If it's Jetson Xavier NX 16GB module, I think you need to upgrade JetPack version.

closed 2023-04-27 by dusty-nv