dusty-nv/jetson-inference #1501: No space left on device

wartek69, 2022-09-22:
Hey,
I'm trying to get the jetson-inference docker to work on the jetson agx xavier.
I've flashed the jetson with a freshly installed jetpack 5.0.2 & followed the docker installation guide.
When running `docker/run.sh` I'm getting the following error:
/usr/local/cuda-11.4/targets/aarch64-linux/lib/libcusolverMg.so.11.2.0.238: no space left on device..
This doesn't really make sense as this is a newly flashed system, I did not put any other packages on it.
When checking the space of the device, I can see the following:
```
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/mmcblk0p1  28768292 18950372   8333532  70% /
```
I've only installed the DeepScene & Cityscapes segmentation models.
Is the docker not supported on the jetson agx xavier dev kit or am I messing something up?
Br
Alex

dusty-nv, 2022-09-22:
It looks like you have ~8GB of free space, but the jetson-inference:r35.1.0 docker image is bigger than that.

I would recommend adding NVMe storage, or for now just building jetson-inference from source (not using docker) as that uses less storage.

wartek69, 2022-09-22:
Thanks for the quick reply!
After following the tips here: https://developer.nvidia.com/blog/minimizing-storage-usage-on-jetson/ I stripped the jetpack version and created enough space for the docker image.
I was at first confused because the same approach seemed to work on Jetpack 4.6 before. But apparently, jetpack 5.0 is ~5Gb larger than Jetpack 4.6 hence the issue.
Many thanks for the assistance!

closed 2022-09-22 by wartek69