dusty-nv/jetson-inference #1503: Questions about training with pascal voc annotations

RoninHunter, 2022-09-27:
I have two questions about training.

If I have an annotated dataset with pascal voc annotations how might I combine that with a downloaded dataset from open image v6 as shown in your example?

For example I want to use some of the classes and their annotations from open image v6 along with some of my own datasets I have personally labeled, How would you combine the two? Do I train with the open v6 dataset and then use the weights and continue training with the other classes and sets or is there a way to combined the two groups of datasets into something I can train on at the same time (OpenImageV6 & MyOwnPascalVOCDatasets)?

My other question is, in your example when you are using your created tractors dataset I see you have an "ImageSets" directory filled with a list of all the images for train, validation, and test. How did you construct those lists?

RoninHunter, 2022-09-29:
I found a really nice utility called oidv6-to-voc that converts the open image v6 csv file to many pascal voc files, however when I tried to train with these files I get the following error:

OSError: missing ImageSet file data/dataFiles/ImageSets/trainval.txt

So I imagen I need to product these ImageSets files. How does one do that if they only have the images and the pascal VOC files?

RoninHunter, 2022-09-30:
I already have my datasets labelled in Pascal VOC format so I don't need to use CVAT. So could you share either a working example of a trainval.txt so I can see the layout or copy a small snippet of the layout so I know what the script output should look like?

dusty-nv, 2022-09-30:
Sure - here's a small snippet from `trainval.txt` from my tractors dataset:

```
20200917-184908
20200917-184919
20200917-184930
20200917-184942
20200917-184956
20200917-185013
20200917-185030
```

These image ID's correlate with the name of the jpg files and the xml files

RoninHunter, 2022-09-30:
[imageFIle-xmlFile] ?

Also how does trainval.txt differ from train.txt and val.txt?

dusty-nv, 2022-09-30:
> [imageFIle-xmlFile] ?

No, those are date-time stamps from my snippet.   Both the XML and JPG files have the same name, just different extensions.   So for example, I have `JPEGImages/20200917-184908.jpg` and `Annotations/20200917-184908.xml`

> Also, how does trainval.txt differ from train.txt and val.txt?

For actual models, the dataset will be split into train/val/test subsets (typically at 80/10/10 or 70/15/15 percent splits)
However for my simple test models, it's fine to just re-use all the data so that they aren't split.   So just create `trainval.txt` with all the image IDs, and then copy that file to `train.txt` `val.txt` `test.txt`

RoninHunter, 2022-09-30:
Yes very good I understand now thank you so much.

RoninHunter, 2022-10-03:
"For actual models, the dataset will be split into train/val/test subsets (typically at 80/10/10 or 70/15/15 percent splits)"

When you are running an actual model how does one set upt training_ssd.py to overlooks the trainval.txt? Looking through the --help I do not see an option to do so. Am I overlooking something?

dusty-nv, 2022-10-03:
train_ssd.py always uses trainval.txt for training and test.txt for validation (I realize this is a misnomer)

You can change this is you want under vision/datasets/voc.py in the source

closed 2023-04-27 by dusty-nv