dusty-nv/jetson-inference #1506: Using converted .engine gives dims error.

arufl, 2022-10-05:
Hello, I was wondering if I could use the .engine file that jetson-inference generates after one run instead of the .onnx. After, I tried it with a custom imagenet, it gives me an error due to the wrong dims.

This is a truncated output of the .onnx model which works fine:
```
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       28
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 4415488
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  3
                -- dim #2  224
                -- dim #3  224
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  2
                -- dim #0  1
                -- dim #1  28
[TRT]    
[TRT]    binding to input 0 input_0  binding index:  0
[TRT]    binding to input 0 input_0  **dims (b=1 c=3 h=224 w=224)** size=602112
[cuda]   cudaAllocMapped 602112 bytes, CPU 0x100ca8200 GPU 0x100ca8200
[TRT]    binding to output 0 output_0  binding index:  1
[TRT]    binding to output 0 output_0  **dims (b=1 c=28 h=1 w=1)** size=112
[cuda]   cudaAllocMapped 112 bytes, CPU 0x100d3b200 GPU 0x100d3b200
[TRT]    
[TRT]    device GPU, resnet18.onnx initialized.
[TRT]    imageNet -- loaded 28 class info entries
[TRT]    imageNet -- resnet18.onnx initialized.
```

And this is the output of the onnx.1.1.8201.GPU.FP16.engine model:

```
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       28
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 4415488
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  3
                -- dim #2  224
                -- dim #3  224
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  2
                -- dim #0  1
                -- dim #1  28
[TRT]    
[TRT]    binding to input 0 input_0  binding index:  0
[TRT]    binding to input 0 input_0  **dims (b=1 c=1 h=3 w=224)** size=2688
[cuda]   cudaAllocMapped 2688 bytes, CPU 0x100ca8200 GPU 0x100ca8200
[TRT]    binding to output 0 output_0  binding index:  1
[TRT]    binding to output 0 output_0  **dims (b=1 c=1 h=28 w=1)** size=112
[cuda]   cudaAllocMapped 112 bytes, CPU 0x100ca8e00 GPU 0x100ca8e00
[TRT]    device GPU, initialized resnet18.onnx.1.1.8201.GPU.FP16.engine
[TRT]    imageNet -- loaded 28 class info entries
[TRT]    imageNet -- didn't load expected number of class descriptions  (28 of 1)
[TRT]    imageNet -- failed to load synset class descriptions  (28 / 28 of 1)
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet
```
As you could see in bold letters, there is a dims mismatch. What can be done to overcome this?

Thanks in advance.

dusty-nv, 2022-10-05:
Hi @arufl, what's the command line or code that you are using to run this?

You shouldn't have to explicitly tell it to run the .engine file - after jetson-inference creates it on the first run, it will automatically detect it and use that instead.

arufl, 2022-10-06:
> Hi @arufl, what's the command line or code that you are using to run this?
`imagenet --model=resnet18.onnx.1.1.8201.GPU.FP16.engine  --labels=labels.txt --input_blob=input_0 --log-level=debug --output_blob=output_0 letras-bin/A/0.jpg`

My objective was to check if I could generate the engine file and just copy it across multiple cloned Nanos.

dusty-nv, 2022-10-06:
> My objective was to check if I could generate the engine file and just copy it across multiple Nanos.

If those Nano's are running the same version of JetPack, yes you should be able to do that.   You can copy the .engine file between those Nano's then.  However, to debug this issue can you try just running imagenet program with the ONNX?   It should automatically find the existing .engine file then

arufl, 2022-10-06:
I did a clean git clone of the repo. However, the input_0 and output_0 dims mismatch is still there. 
```
[video]  created imageLoader from file:///home/user/ejemplos/100.jpg
------------------------------------------------
imageLoader video options:
------------------------------------------------
  -- URI: file:///home/user/ejemplos/100.jpg
     - protocol:  file
     - location:  100.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     input
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
imagenet:  failed to create output stream
[cuda]   cudaAllocMapped 65536 bytes, CPU 0x100c60000 GPU 0x100c60000
[cuda]   cudaAllocMapped 262144 bytes, CPU 0x100c60000 GPU 0x100c60000
[cuda]   cudaAllocMapped 16384 bytes, CPU 0x100ca0000 GPU 0x100ca0000
[cuda]   cudaAllocMapped 16384 bytes, CPU 0x100ca4000 GPU 0x100ca4000

imageNet -- loading classification network model from:
         -- prototxt     (null)
         -- model        resnet18.onnx.1.1.8201.GPU.FP16.engine
         -- class_labels labels.txt
         -- input_blob   'input_0'
         -- output_blob  'output_0'
         -- batch_size   1

[TRT]    TensorRT version 8.2.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - engine  (extension '.engine')
[TRT]    loading network plan from engine cache... resnet18.onnx.1.1.8201.GPU.FP16.engine
[TRT]    [MemUsageChange] Init CUDA: CPU +224, GPU +0, now: CPU 296, GPU 2766 (MiB)
[TRT]    Loaded engine size: 45 MiB
[TRT]    Using cublas as a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +159, GPU +162, now: CPU 455, GPU 2928 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +240, GPU +237, now: CPU 695, GPU 3165 (MiB)
[TRT]    Deserialization required 3426865 microseconds.
[TRT]    [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +45, now: CPU 0, GPU 45 (MiB)
[TRT]    Using cublas as a tactic source
[TRT]    [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU +0, now: CPU 695, GPU 3165 (MiB)
[TRT]    Using cuDNN as a tactic source
[TRT]    [MemUsageChange] Init cuDNN: CPU +0, GPU +0, now: CPU 695, GPU 3165 (MiB)
[TRT]    Total per-runner device persistent memory is 42887680
[TRT]    Total per-runner host persistent memory is 27296
[TRT]    Allocated activation device memory of size 2408448
[TRT]    [MemUsageChange] TensorRT-managed allocation in IExecutionContext creation: CPU +0, GPU +43, now: CPU 0, GPU 88 (MiB)
[TRT]
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       27
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 2408448
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  3
                -- dim #2  224
                -- dim #3  224
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  2
                -- dim #0  1
                -- dim #1  28
[TRT]
[TRT]    binding to input 0 input_0  binding index:  0
[TRT]    binding to input 0 input_0  dims (b=1 c=1 h=3 w=224) size=602112
[TRT]    binding to output 0 output_0  binding index:  1
[TRT]    binding to output 0 output_0  dims (b=1 c=1 h=28 w=1) size=112
[cuda]   cudaAllocMapped 112 bytes, CPU 0x100ca8200 GPU 0x100ca8200
[TRT]    device GPU, initialized resnet18.onnx.1.1.8201.GPU.FP16.engine
[TRT]    loaded 28 class labels
[TRT]    didn't load expected number of class descriptions  (28 of 1)
[TRT]    imageNet -- failed to load synset class descriptions  (28 / 28 of 1)
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet
```

dusty-nv, 2022-10-10:
Hi @arufl, please run imagenet with `--model=resnet18.onnx` instead of `--model=resnet18.onnx.1.1.8201.GPU.FP16.engine`

It will still use the cached .engine file that it finds, however point it at the .onnx for it to run correctly

arufl, 2022-10-10:
Okay. Thanks for your support.

closed 2022-10-10 by arufl