dusty-nv/jetson-inference #1508: jetson nx --- utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device

whaosoft, 2022-10-14:
Jetson NX (jetpack 4.6.2)
TensorRT ：8.2.1.8
CUDA ：10.2
CUDNN ：8.2.1
Same error when running official onnx model: /usr/src/tensorrt/bin/trtexec --onnx=./data/resnet50/ResNet50.onnx

[utils.cpp::checkMemLimit::380] Error Code 2: Internal Error (Assertion upperBound != 0 failed. Unknown embedded device detected. Please update the table with the entry: {{1794, 6, 16}, 12660},)

dusty-nv, 2022-10-14:
Hi @whaosoft, what variant of Jetson Xavier NX are you running?   Is it Jetson Xavier NX 16GB?   If so, I think you need to upgrade your version of JetPack / TensorRT

whaosoft, 2022-10-15:
> Hi @whaosoft, what variant of Jetson Xavier NX are you running? Is it Jetson Xavier NX 16GB? If so, I think you need to upgrade your version of JetPack / TensorRT

 My version is Jetson Xavier NX emmc 16GB + ssd / china dev kit
 Are you sure upgrade to Jetpack 5.0 version? 
I've tried to run many many times, it drive me mad...
I've heard that jetpack 4.6.0 is okay , only 4.6.2 hava bug? Do you think so?

whaosoft, 2022-10-15:
@dusty-nv 

> Hi @whaosoft, what variant of Jetson Xavier NX are you running? Is it Jetson Xavier NX 16GB? If so, I think you need to upgrade your version of JetPack / TensorRT
~~~
My version is Jetson Xavier NX emmc 16GB + ssd / china dev kit
Are you sure upgrade to Jetpack 5.0 version?
I've tried to run many many times, it drive me mad...
I've heard that jetpack 4.6.0 is okay , only 4.6.2 hava bug? Do you think so?