dusty-nv/jetson-inference #1510: Would it be possible to derive translation and rotation of the camera

neilyoung, 2022-10-24:
...with one of these AI methods? E.g. object detection/segmentation on people, find distance and rotation in 6DoF

dusty-nv, 2022-10-24:
Hi @neilyoung, do you mean the frame-to-frame egomotion of the camera, or the orientation of the camera relative to a detected object?   I haven't done that before but maybe there is some stuff in OpenCV that can calculate it.

neilyoung, 2022-10-24:
Hi Dusty. I see, you haven't changed your habits to be fast and responsive, not even over pandemic and war. I really like that.

Yes, you got it. Simple task: Visual "follow me" of a person "on camera" from a drone. I had some first success with AI Pose detection and "solvePnP" (https://learnopencv.com/tag/solvepnp/) from OpenCV, but usually the drone is not necessarily having a closeup of the face while doing that. Other than face pose points there is not much constant on a human body, which would allow to derive fix world coordinates in order to match the image coordinates in order to derive the drone's (camera's) pose (translation and rotation). 

I need that in order to derive the steering commands from the drone. I already did that with OpenCV means and Aruco codes for precise landing, but I doubt anybody would ware Aruco codes just be allowing me to follow him :))

dusty-nv, 2022-10-24:
Thanks Neil, always happy to help - although this particular area may not be my forte, sorry about that.   

In the past when I had played around with "follow me", it was on a ground robot with a simple command loop that aimed to keep the tracked object in the center of the frame.   However I can imagine with a drone how you would need to translate that to world coordinates to inform the flight planner.

If it's helpful, you may want to see if the mono depth DNN from jetson-inference can provide you with an useful info.  Otherwise if you have a stereo camera, doing actual stereo depth mapping would be beneficial I imagine.

neilyoung, 2022-10-24:
That sounds interesting. I also already thought about that "simple" approach: Identify the initial bounding box (e.g. pick a person out of bunch to follow only him), apply object detection and segmentation (to assign him a number once he is mixing with others), trying to match the initial bounding box first. Then have a PID controller or such for the yaw, x and y rotations with somehow all try to determine the steering values (I'm also in charge for steering the drone, so I know what to do in order to move left/right/forward/back/up/down and how to rotate over yaw), which keep "that particular bounding box" in the center of the screen :)

neilyoung, 2022-10-25:
What about the monodepth thing? What is the detection range? Would a combination of object detection and depths map feasible?

dusty-nv, 2022-10-26:
The monodepth stuff is here - https://github.com/dusty-nv/jetson-inference/blob/master/docs/depthnet.md

Unlike stereo, it's relative not absolute, so it scales depending on the scene.  You can attempt to calibrate it for your camera but YMMV.  I haven't tried doing 3D object detection with it, a 6-DoF object detection model (not in this repo) might be better for that.

neilyoung, 2022-10-26:
Do you have an example for such a 6dof model? A name?

closed 2023-04-27 by dusty-nv