dusty-nv/jetson-inference #1511: importing jetson.inference is depreciated.  Please 'import jetson.inference' instead

Michael-EVOL, 2022-10-30:
Following Hello AI world on jetson nano 2GB,

Trying to write the my-recognition.py example in python, and am told that: 
"importing jetson.inference is depreciated.  Please 'import jetson_inference' instead"
and
"importing jetson.utils is depreciated.  Please 'import jetson_utils' instead"

When I update the imports, I also update the call to load an image
from => jetson.utils.loadimage() 
to     => jetson_utils.loadimage() 

but receive the error message:
"jetson_utils has no attribute 'loadimage'"

Where might I find the updated attributes to call for the new import?

dusty-nv, 2022-10-31:
Hi @Michael-EVOL, it's `jetson_utils.loadImage()`  (note the capitalization)

Michael-EVOL, 2022-10-31:
Aha, that is what I thought it was going to be a capitalized version of the one I was using.  

Also I found that the function imagenet() became imageNet() in the new package.

Thank you @dusty-nv very much for your help with this issues.

dusty-nv, 2022-10-31:
No problem - the function names haven't changed (i.e. it was always `imageNet`)

closed 2023-04-27 by dusty-nv