dusty-nv/jetson-inference #1512: Omitting detections with bbox area greater than specified value

RoninHunter, 2022-11-03:
I am trying to omit any detections that would be drawn to the screen using detectnet.py that have a bbox greater than a specified value. I don't want to draw these detections on screen. Would I do this in PydetectNet.cpp?

Again I am trying to omit any detections drawn to the screen above a specified area value. 

Any help would be amazing.

Thank you

dusty-nv, 2022-11-07:
Hi @RoninHunter, what you would do is skip the overlay during `detectNet.Detect(img, overlay='none')`.  Then manipulate the list of detections that it returns, removing the ones that you don't want.  Then you can call `detectNet.Overlay(img, detections, overlay='box,labels,conf')` to perform the overlay on your list of detections.

RoninHunter, 2022-11-08:
Thank you so much dusty. Would you know where can I find the list of members for the object the detectNet.Detect is returning so that I can manipulate that list?

dusty-nv, 2022-11-08:
@RoninHunter you can find them on this page under the `Detection = <type 'jetson.inference.detectNet.Detection'>` section:

https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/jetson.inference.html#detectNet

```
Detection = <type 'jetson.inference.detectNet.Detection'>
Object Detection Result
 
----------------------------------------------------------------------
Data descriptors defined here:
 
Area
    Area of bounding box
 
Bottom
    Bottom bounding box coordinate
 
Center
    Center (x,y) coordinate of bounding box
 
ClassID
    Class index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/__builtin__.html#object)
 
Confidence
    Confidence value of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/__builtin__.html#object)
 
Height
    Height of bounding box
 
Instance
    Instance index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/__builtin__.html#object)
 
Left
    Left bounding box coordinate
 
Right
    Right bounding box coordinate
 
Top
    Top bounding box coordinate
 
Width
     Width of bounding box
```

closed 2023-04-27 by dusty-nv