dusty-nv/jetson-inference #1517: cudaCrop() failed to get input/output image pointers (should be cudaImage)

Gensight, 2022-11-07:
I am trying to just crop an image and I get this error:
cudaCrop() failed to get input/output image pointers (should be cudaImage)
I am using basic camera.Capture() to get the initial image and cuda to allocate memory for the destination image. Here is my code:

                initalImage = camera.Capture()

                cropBorder=((initalImage.width-1024)/2,(initalImage.height-768)/2)
                crop_roi = (cropBorder[0], cropBorder[1], 1024 + cropBorder[0], 768 + cropBorder[1])

                activeImage = jetson.utils.cudaAllocMapped(1024, 768, format=initalImage.format)
---> Here is where I get the error message:
                jetson.utils.cudaCrop(initalImage, activeImage, crop_roi)

                detections = net.Detect(activeImage)

Please help, my project depends on this working.

Gensight, 2022-11-07:
The camera is acquired with this line:
camera = jetson.utils.videoSource("csi://0")

dusty-nv, 2022-11-07:
Hi @Gensight, can you try `jetson.utils.cudaAllocMapped(width=1024, height=768, format=initalImage.format)` instead?

Gensight, 2022-11-07:
Thank you so much. I am getting an invalid ROI now so I'll work on that.

dusty-nv, 2022-11-07:
OK, that's good progress - I would try either reducing or increasing your ROI by 1px

Gensight, 2022-11-08:
Turns out that jetson.utils.videoSource("csi://0") only runs at 1280x720 and I was trying to crop using 768. I have to figure out how to get the correct resolution I need with videoSource() but I am good for now. Thank you again, you are great!

dusty-nv, 2022-11-08:
for CSI cameras, videoSource will just support the modes the camera supports. You could pick the closest mode the camera supports with something like this:  `jetson.utils.videoSource("csi://0", argv=['--input-width=1024', '--input-height=768'])`    (these modes should be printed out near the beginning of the terminal log).  Otherwise, you could rescale with [`jetson.utils.cudaResize()`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#resizing)

Gensight, 2022-11-08:
Yes, I found the --input-width/height parameters and they work great.

closed 2023-04-27 by dusty-nv