dusty-nv/jetson-inference #1518: listDir('/usr/bin/image/peds_0.jpg') - found no matches

venugopalreddt, 2022-11-14:
Hey @dusty-nv  
So I was trying to process a couple of test images from console using `detectnet.py` and I ran this code:
`venug@venug-desktop:~/jetson-inference/build/aarch64/bin$ ./detectnet-console.py image/peds_0.jpg output_0.jpg`

and this was the output given:

`listDir('/usr/bin/image/peds_0.jpg') - found no matches
[image]  imageLoader -- failed to find 'image/peds_0.jpg'
Traceback (most recent call last):
  File "./detectnet-console.py", line 51, in <module>
    input = videoSource(args.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
`
I checked thoroughly and the images peds_0 **was** in the folder, so I dont know whats wrong. 
Can you or anyone else please help me with this? Thank you in advance :smile:

dusty-nv, 2022-11-14:
Hi @venugopalreddt, I think there was a small typo in your command and you should try running this instead:

``` bash
./detectnet-console.py images/peds_0.jpg output_0.jpg
```

venugopalreddt, 2022-11-26:
Hi @dusty-nv , wow that was a silly mistake lol. Sorry for disturbing you for that.  Oh and btw I suppose the images in the "images" aren't useful right? Like even if I delete them the object recognition should work or am I wrong? Thanks lol

dusty-nv, 2022-11-28:
> Oh and btw I suppose the images in the "images" aren't useful right? Like even if I delete them the object recognition should work or am I wrong?

No problem, yes those images are just for testing purposes really, deleting them will not affect the function of the models.