dusty-nv/jetson-inference #1519: --batch_size=2

CapBarbossa, 2022-11-15:
When inference, with  default value 1 of batch_size, everything works just fine. With a value greater than 1, say 2, then, `Detect` methed or Process method cant work.
Take posenet fro example:
```python
net = poseNet('resnet18-body', ['--batch_size=2'], 0.15)

dummy = np.random.randint(0,255,[2,640,480,3],dtype=np.uint8)
net.Process(jetson_utils.cudaFromNumpy(dummy))
SystemError: <built-in method Process of jetson.inference.poseNet object at 0x7f
5f23f910> returned NULL without setting an error
```
Why? You designers of jetson-inference are all fucking idots? Aren't Batch size greater than 1 more normal and universal? You suckers.

dusty-nv, 2022-11-15:
@CapBarbossa batch size > 1 isn't fully implemented - jetson-inference is mostly intended for single-stream / low-latency and the Hello AI World tutorial.   For high-throughput multi-stream production applications I'd recommend using DeepStream.   Please keep your comments civil.

CapBarbossa, 2022-11-21:
Sorry. And again I am wrong from the very begining. I will try DeepStream. Thanks for your attribute to Open Source and I am   ashamed of what I said, but I am NOT regret it. "isn't fully implemented" is reasonabe, but could you please specify in an outstanding way, telling us this information. Currently in this function help info, it stills holds the batch_size option, it readlly does. 

Best regards
yiting

dusty-nv, 2022-11-21:
> Currently in this function help info, it stills holds the batch_size option, it readlly does.

Thanks @CapBarbossa, you are right about that - I've made a note in my TODO list to remove batch-size from the `--help`

dusty-nv, 2023-04-14:
@jet-c-21 yes, single-frame inference