dusty-nv/jetson-inference #1520: How to import jetson.inference in Jupyter(jetbot)

PhotloJ, 2022-11-15:
I use jetbot to detect object but I have to use it via jupyter. How can I import this?

dusty-nv, 2022-11-15:
Hi @PhotloJ, you would need to install jetson-inference into the jetbot's container, or change the jetbot's base container to the jetson-inference container.  You should then be able to `import jetson.inference`

PhotloJ, 2022-11-23:
How can I install jetson-inference into the jetbot's container. I'm new for this. Please guide me. Thank you so much.

dusty-nv, 2022-11-23:
I would look at the jetson-inference dockerfile to see how it's done:   https://github.com/dusty-nv/jetson-inference/blob/master/Dockerfile

PhotloJ, 2022-11-28:
I tried to follow your guide but I can't. May be my skill isn't much. Some one please help.