dusty-nv/jetson-inference #1521: Low FPS in video-viewer using v4l2src

poett1, 2022-11-15:
Hello Dusty,
I'm building an application based on jetson-utils/jetson-inference and I can't get framerates higher than 26 FPS from a V4L2 USB Cam (max. 60 FPS supported). In addition to this, `[gstreamer] gstreamer message qos ==> v4l2src0`  is printed in the terminal several times a second.
To reproduce this I tried `video-viewer` for the V4L2 cam and also a nvargus supported camera. While running `video-viewer v4l2:///dev/video0` I also noticed a lower framerate compared to `video-viewer csi://0` (around 26 FPS for V4L2, 30 FPS for CSI/argus)  and also got the gstreamer qos message. Processing should therefore be limited in the input pipeline and not in the output pipeline. But because it's only 720p by default, I wouldn't assume any bottlenecks yet. 
To double check this in a pure gstreamer pipeline, i executed: `gst-launch-1.0 v4l2src device=/dev/video0 ! 'video/x-raw,width=1280,height=720' ! videoconvert ! fpsdisplaysink text-overlay=0 video-sink=xvimagesink -v` and this reaches full 60 FPS from the V4L2 camera. Since `video-viewer` outputs the same input caps as this pipeline (apart from appsink vs fpsdisplaysink) and CSI input also gets higher framerates, i can't explain this reduced framerate. 

Any hints? 

Thanks a lot!

dusty-nv, 2022-11-15:
Hi @poett1, which Jetson and version of JetPack-L4T are you using?  My guess is that since this is running in RAW mode, it's related to overhead from appsink since I'm not able to use NVMM (zero-copy) memory in RAW mode.   Do you have a display attached?  It might get better performance without.

poett1, 2022-11-15:
I'm using a Jetson Xavier NX Dev Kit with Jetpack 5.0.2.
I already tried the input pipeline without any processing/display output like this:
```
while( true )
	{	
        // capture next image
		if( !input->Capture(&image, 1000) )
		{
			std::cerr << "Error: failed to capture video frame" << std::endl;
			continue;
		}
		// log timing
		double cur = (double) cv::getTickCount();
		double delta = (cur - prev) / cv::getTickFrequency();
		std::cout<<"FPS=" << 1.0/delta << std::endl;
		prev=cur;
    }
```
And it unfortunately is reaching exactly the same FPS.

poett1, 2022-11-22:
I had a few more thoughts on this topic. If the limiting factor is only the lack of NVMM memory in our pipeline passed to appsink, there could be 2 potential solutions (untested):
1. Adding something like **nvvidconv**, that converts the video to NVMM before appsink:
`v4l2src device=/dev/video0 do-timestamp=true ! 'video/x-raw, format=(string)UYVY, width=(int)1280, height=(int)720' !  nvvidconv ! 'video/x-raw(memory:NVMM), format=NV12' ! appsink name=mysink`
2. Exchanging **v4l2src** by **nvv4l2camerasrc**, that outputs NVMM:
`nvv4l2camerasrc device=/dev/video0 ! 'video/x-raw(memory:NVMM),format=UYVY,width=640,height=480,framerate=30/1' ! nvvidconv ! 'video/x-raw(memory:NVMM),format=NV12' ! appsink name=mysink`

What's your take on this, @dusty-nv? Might that help?

poett1, 2023-01-05:
I was able to solve this by using `nvv4l2camerasrc` instead of `v4l2src`. 

Output for the Input pipeline looks like this: 
`[gstreamer] nvv4l2camerasrc device=/dev/video0 ! video/x-raw(memory:NVMM), format=(string)UYVY, interlace-mode=progressive, framerate=60/1, width=(int)1280, height=(int)720 ! nvvidconv flip-method=0 ! video/x-raw ! appsink name=mysink`

closed 2023-04-27 by dusty-nv