dusty-nv/jetson-inference #1523: Hwo to make install jetson-inference to a python env created by conda

pchly, 2022-11-21:
The Python bindings for the [jetson.inference] and [jetson.utils] modules also get installed during the sudo make install step under /usr/lib/python*/dist-packages/. But i want install jetson-inference in my own python env,this python env is created by miniforge.

dusty-nv, 2022-11-21:
For now you could just change the python install path here:

* https://github.com/dusty-nv/jetson-inference/blob/9ee9a950a80fbc7597d7e78a7ba0a282e85fae78/python/bindings/CMakeLists.txt#L76
* https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/python/bindings/CMakeLists.txt#L76

If you have a programatic way of detecting this path in CMake, taking into account python virtualenvs/conda/miniforge/ect, that would be very handy to know

pchly, 2022-11-22:
Thank you for your answer. I set up according to your instructions, but still found the following problems：
1.Only folder about jetson-inference is installed, but folder about jetson-utils is not successfully installed;

pchly, 2022-11-22:
I found the reason. I didn't set the jetson-information/utils/python/bindings/CMakelist.txt

closed 2023-04-27 by dusty-nv