dusty-nv/jetson-inference #1524: change the version of python

pchly, 2022-11-22:
when i change the version of python in jetson-inference/python/CmakeList.txt to 3.6,as shown in the following code:
```
if(LSB_RELEASE_CODENAME MATCHES "focal")
	set(PYTHON_BINDING_VERSIONS 3.6)
else()
	set(PYTHON_BINDING_VERSIONS 2.7 3.6 3.7)
endif()
```
but I got an error as follows:
`ImportError: /home/pch/miniforge-pypy3/envs/tensorrt_py36/bin/../lib/libffi.so.7: version `LIBFFI_BASE_7.0' not found (required by /lib/aarch64-linux-gnu/libgobject-2.0.so.0)`

dusty-nv, 2022-11-23:
Yes, and if you were ever to want to use PyTorch or TensorFlow, we only build those for Python 3.8 on Ubuntu 20.04.  So I'd recommend switching to Python 3.8 since you are on 20.04

pchly, 2022-11-23:
OK, thank you for your answer. I wanted to try to compile other versions. I accept your suggestion and use python 3.8 in ubuntu20.04

closed 2023-04-27 by dusty-nv