dusty-nv/jetson-inference #1527: Cannot train ssd2-lite models

arufl, 2022-11-30:
Hello. I am following the pytorch-ssd.md guide to train the fruit detector and I would want to train a mb2-ssd-lite net.
` python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=120 --epochs=30 --use-cuda=True --net=mb2-ssd-lite`
However, I got the following error:

```
Traceback (most recent call last):
  File "train_ssd.py", line 361, in <module>
    net.init_from_pretrained_ssd(args.pretrained_ssd)
  File "/home/usuario/AI/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 132, in init_from_pretrained_ssd
    self.load_state_dict(model_dict)
  File "/home/usuario/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1482, in load_state_dict
    raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
RuntimeError: Error(s) in loading state_dict for SSD:
        Unexpected key(s) in state_dict: "base_net.1.0.weight", "base_net.1.1.weight", "base_net.1.1.bias", "base_net.1.1.running_mean", "base_net.1.1.running_var", "base_net.1.3.weight", "base_net.1.4.weight", "base_net.1.4.bias", "base_net.1.4.running_mean", "base_net.1.4.running_var", "base_net.2.0.weight", "base_net.2.1.weight", "base_net.2.1.bias", "base_net.2.1.running_mean", "base_net.2.1.running_var", "base_net.2.3.weight", "base_net.2.4.weight", "base_net.2.4.bias", "base_net.2.4.running_mean", "base_net.2.4.running_var", "base_net.3.0.weight", "base_net.3.1.weight", "base_net.3.1.bias", "base_net.3.1.running_mean", "base_net.3.1.running_var", "base_net.3.3.weight", "base_net.3.4.weight", "base_net.3.4.bias", "base_net.3.4.running_mean", "base_net.3.4.running_var", "base_net.4.0.weight", "base_net.4.1.weight", "base_net.4.1.bias", "base_net.4.1.running_mean", "base_net.4.1.running_var", "base_net.4.3.weight", "base_net.4.4.weight", "base_net.4.4.bias", "base_net.4.4.running_mean", "base_net.4.4.running_var", "base_net.5.0.weight", "base_net.5.1.weight", "base_net.5.1.bias", "base_net.5.1.running_mean", "base_net.5.1.running_var", "base_net.5.3.weight", "base_net.5.4.weight", "base_net.5.4.bias", "base_net.5.4.running_mean", "base_net.5.4.running_var", "base_net.6.0.weight", "base_net.6.1.weight", "base_net.6.1.bias", "base_net.6.1.running_mean", "base_net.6.1.running_var", "base_net.6.3.weight", "base_net.6.4.weight", "base_net.6.4.bias", "base_net.6.4.running_mean", "base_net.6.4.running_var", "base_net.7.0.weight", "base_net.7.1.weight", "base_net.7.1.bias", "base_net.7.1.running_mean", "base_net.7.1.running_var", "base_net.7.3.weight", "base_net.7.4.weight", "base_net.7.4.bias", "base_net.7.4.running_mean", "base_net.7.4.running_var", "base_net.8.0.weight", "base_net.8.1.weight", "base_net.8.1.bias", "base_net.8.1.running_mean", "base_net.8.1.running_var", "base_net.8.3.weight", "base_net.8.4.weight", "base_net.8.4.bias", "base_net.8.4.running_mean", "base_net.8.4.running_var", "base_net.9.0.weight", "base_net.9.1.weight", "base_net.9.1.bias", "base_net.9.1.running_mean", "base_net.9.1.running_var", "base_net.9.3.weight", "base_net.9.4.weight", "base_net.9.4.bias", "base_net.9.4.running_mean", "base_net.9.4.running_var", "base_net.10.0.weight", "base_net.10.1.weight", "base_net.10.1.bias", "base_net.10.1.running_mean", "base_net.10.1.running_var", "base_net.10.3.weight", "base_net.10.4.weight", "base_net.10.4.bias", "base_net.10.4.running_mean", "base_net.10.4.running_var", "base_net.11.0.weight", "base_net.11.1.weight", "base_net.11.1.bias", "base_net.11.1.running_mean", "base_net.11.1.running_var", "base_net.11.3.weight", "base_net.11.4.weight", "base_net.11.4.bias", "base_net.11.4.running_mean", "base_net.11.4.running_var", "base_net.12.0.weight", "base_net.12.1.weight", "base_net.12.1.bias", "base_net.12.1.running_mean", "base_net.12.1.running_var", "base_net.12.3.weight", "base_net.12.4.weight", "base_net.12.4.bias", "base_net.12.4.running_mean", "base_net.12.4.running_var", "base_net.13.0.weight", "base_net.13.1.weight", "base_net.13.1.bias", "base_net.13.1.running_mean", "base_net.13.1.running_var", "base_net.13.3.weight", "base_net.13.4.weight", "base_net.13.4.bias", "base_net.13.4.running_mean", "base_net.13.4.running_var", "extras.0.0.weight", "extras.0.0.bias", "extras.0.2.weight", "extras.0.2.bias", "extras.1.0.weight", "extras.1.0.bias", "extras.1.2.weight", "extras.1.2.bias", "extras.2.0.weight", "extras.2.0.bias", "extras.2.2.weight", "extras.2.2.bias", "extras.3.0.weight", "extras.3.0.bias", "extras.3.2.weight", "extras.3.2.bias".
```
I would be grateful if someone could guide me to solve this issue.
Thanks in advance.

dusty-nv, 2022-11-30:
@arufl you could look to the upstream repo https://github.com/qfgaohao/pytorch-ssd
ssd-mobilenet-v1 is the only one I have tested/working with the PyTorch -> ONNX -> TensorRT workflow

arufl, 2022-12-01:
I also failed to make it run due to a shape mismatch during onnx export. I will stick to ssd1 for the moment.

Thanks anyway.

closed 2022-12-01 by arufl