dusty-nv/jetson-inference #1528: Building the project from source

mohitg0292, 2022-12-03:
**Jetson Nano developer kit Specs:**
Jetson Nano 4GB kit with 16 Memory card.
L4T version 32.6.1 [JetPack 4.6]
Ubuntu 18.04.5

**Errors:**
1. I'm getting errors in the command "cmake ../"
[image]
[image]
[image]

2. In the step installpytorch.sh I'm getting Pytorch for Python 3.6 as shown in the screenshot below.
[image]
I tried to install pytorch for python 3.6 but it didn't complete because of memory issues.

3. Make commands also showed errors.
[image]

Please suggest some solution.
Thank you.

dusty-nv, 2022-12-04:
Hi Mohit, can you try running cmake again?  It should skip the dialogs this time

mohitg0292, 2022-12-04:
Hi Dustin,
I'm getting the same errors after running the cmake command.

dusty-nv, 2022-12-04:
Did you run cmake ../ from the build directory and did it complete successfully?

mohitg0292, 2022-12-04:
Yes I ran the the command "cmake ../" from build directory. But it completed with errors which I put screenshot above.

dusty-nv, 2022-12-04:
OK, I see.  It appears that you do not have CUDA toolkit installed on your Jetson.  Did you use the SD card image or SDK Manager to flash it?

I believe you should able to do 'sudo apt-get install cuda-toolkit-*'

mohitg0292, 2022-12-04:
I have just run the command and it's still running. I'm not sure how the image is loaded into SD card. Could you please share if there's a way to check if the Jetson Developer Kit image is loaded into the memory card?
Also is 16GB SD card enough?

dusty-nv, 2022-12-04:
16GB is small yes, so I would get a bigger one eventually

mohitg0292, 2022-12-04:
The command you mentioned got incompleted because of space issues.
[image]

I would also increase the memory size as you suggested.

mohitg0292, 2022-12-10:
Hi!

So I just realized that the Ubuntu OS is installed in internal memory which is around 15GB. I'm not able to use SD card, it's not showing even with "df -Th" command. Not sure why. 
Do you think this project can only be run in SD card? Because the internal memory got full by trying different methods to fix the errors and now even after removing the project, the memory is still full.

Mohit

dusty-nv, 2022-12-10:
> I'm not able to use SD card, it's not showing even with "df -Th" command. Not sure why.

Hi @mohitg0292, I would try using the ubuntu GUI disk tool and make sure your SD card is formatted as ext3/ext4 and is setup to auto-mount.

> Do you think this project can only be run in SD card? Because the internal memory got full by trying different methods to fix the errors and now even after removing the project, the memory is still full.

I've run it before in 16GB + JetPack, but yes JetPack already takes up a lot of space.

ligaz, 2022-12-10:
@mohitg0292 You can [this guide](https://nvidia-ai-iot.github.io/jetson-min-disk/intro.html) from Nvidia on how to minimizing Jetson disk usage. Hope this helps.

closed 2023-04-27 by dusty-nv