dusty-nv/jetson-inference #1529: ISAAC-ROS overlap

MikeGroenewald, 2022-12-04:
Hi @dusty-nv, first of all, thanks for all your hard work in assembling the jetson group of repositories, they are great and have dropped the barrier to entry with this all this stuff for newcomers like myself, so thanks for that.

I have been looking at both your repositories as well as the ISAAC-ROS group of repositories and am trying to understand the benefits and tradeoffs of both groups as there appears to be quite a bit of overlap. I was hoping you could maybe elaborate on when to use which please? Hope you don't mind as this is not technically an jetson-inference issue, but I could do with the guidance.

behretv, 2023-03-09:
From the repositories I can see, that the docker images from this repo are much smaller in size and the images have a tag for each L4T version. Maybe the Issac images are compatible with every L4T version. Have you tested the Isaac images with different L4T versions?

MikeGroenewald, 2023-03-10:
To be fair, I have only recently picked this up again and I think I am going to try and get the Jetson-Inference stuff working first as there seems to be more support surrounding this project and dusty seems to be very active in the community regarding this. Once I can get that working using my own container I am then going to try and get the deep learning part of this working using my own ROS nodes and containers etc. Once I am happy with that, I will look into the ISAAC-ROS stuff. I am trying to run this on x86 with a GPU so will see how it goes.

dusty-nv, 2023-03-10:
Hi @MikeGroenewald @behretv my containers are "vanilla" ROS/ROS2 containers, whereas Isaac ROS has support for the NVIDIA Isaac nodes in them too.  They had forked my dockerfiles to adapt them for their own purposes, which is an intended use-case of this repo since all my dockerfiles and build scripts are open-source.

closed 2023-04-27 by dusty-nv