dusty-nv/jetson-inference #1530: Jetson inference on Jetpack 5.0.2

SachaMk2, 2022-12-04:
Hello,

I would like to know if we can use jetson inference on Jetpack 5.0?

If so, I saw that an unofficial xubuntu 20.04 version has been created for Jetson nano, would it be possible to run jetson inference on it.

here is the link to the xubuntu 20.04 image for jetson nano: https://forums.developer.nvidia.com/t/xubuntu-20-04-focal-fossa-l4t-r32-3-1-custom-image-for-the-jetson-nano/121768

Thanks in advance, Sincerely,

Sacha

dusty-nv, 2022-12-05:
Hi @SachaMk2, I don't have experience running xubuntu 20.04, but I guess you could always try building jetson-inference on it.  I don't officially support it though.

SachaMk2, 2022-12-05:
Hi, 

Thanks for your answer.
Even if you don't know xubuntu 20.04 for Jetson Nano, do you see some possible issues that I'd probably get if I run jetson_interference on it?

Thanks you in advance,

Sach

dusty-nv, 2022-12-05:
Hi Sach, the issues I could see are less to do with jetson-inference and more to do if the CUDA/cuDNN/TensorRT libraries operate correctly under this third-party xubuntu 20.04 distribution.   If you have that GPU stuff working okay independently, then you could give jetson-inference a try but no guarantees it will work for the above reasons.

SachaMk2, 2022-12-05:
Okay thanks you. I'll give it a try and come back to you if you're interested about the results. 

Sach

JereKnuutinen, 2023-02-02:
@SachaMk2 Hi, did you try it with Jetpack 5.0.

closed 2023-04-27 by dusty-nv