dusty-nv/jetson-inference #1532: How to measure a performance using confusion matrix?

horyan971211, 2022-12-07:
Hi all. I currently running train.py which is from jetson inference. I wish to measure the network performance by using a confusion matrix. 
How am I going to do that? Any recommend for coding?

dusty-nv, 2022-12-07:
Hi @horyan971211 it would need to be with a model that you trained in PyTorch/TensorFlow/ect.   jetson-inference runtime library is for inferencing only and doesn't load the groundtruth dataset, so it wouldn't have that data to compute the confusion matrix.

dusty-nv, 2022-12-07:
Sure, you could add scikit-learn to this PyTorch train.py script for classification models for example:  https://github.com/dusty-nv/pytorch-classification/blob/dd2548357acb46e376543a3475942075f4a5ce88/train.py

horyan971211, 2022-12-07:
> Sure, you could add scikit-learn to this PyTorch train.py script for classification models for example: https://github.com/dusty-nv/pytorch-classification/blob/dd2548357acb46e376543a3475942075f4a5ce88/train.py

One more question. Did the jetson inference is for deep learning or transfer learning?

dusty-nv, 2022-12-07:
By default that train.py script will use transfer learning - it starts with models that were pre-trained on the 1000-class ImageNet dataset.  However you can configure it so that it trains from scratch, but it will take longer to train then.

horyan971211, 2022-12-25:
> By default that train.py script will use transfer learning - it starts with models that were pre-trained on the 1000-class ImageNet dataset. However you can configure it so that it trains from scratch, but it will take longer to train then.

> Sure, you could add scikit-learn to this PyTorch train.py script for classification models for example: https://github.com/dusty-nv/pytorch-classification/blob/dd2548357acb46e376543a3475942075f4a5ce88/train.py

Can you please provide example?

dusty-nv, 2022-12-26:
Sorry, I haven't done it.   https://forums.developer.nvidia.com/t/how-to-measure-acc-loss-f1-etc-on-jetson-nano/237878/2

closed 2023-04-27 by dusty-nv