dusty-nv/jetson-inference #1534: SyntaxError: future feature annotations is not defined

markov24, 2022-12-14:
Whenever I run the train_ssd.py script, I keep getting this error:

  from PIL import Image
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/Pillow-9.3.0-py3.6-linux-aarch64.egg/PIL/Image.py", line 52, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 951, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 894, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1157, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1131, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1112, in _legacy_get_spec
  File "<frozen importlib._bootstrap>", line 441, in spec_from_loader
  File "<frozen importlib._bootstrap_external>", line 544, in spec_from_file_location
  File "/usr/local/lib/python3.6/dist-packages/Pillow-9.3.0-py3.6-linux-aarch64.egg/PIL/_deprecate.py", line 1
SyntaxError: future feature annotations is not defined

I'm not too sure what might be causing this, I thought it was something to do with some python libraries missing, I reran - pip3 install -v -r requirements.txt - but I still keep getting the error.

dusty-nv, 2022-12-14:
Hi @markov24, can you try running this:   `pip3 install 'pillow<9'`

markov24, 2022-12-16:
@dusty-nv Thank you! This solved the issue. I'm curious though, was this something that was supposed to be included in JetPack?

dusty-nv, 2022-12-16:
No, it's because the latest version of pillow only supports Python 3.7+ but this wasn't properly encoded in their wheel versions.  So it'll still install on Python 3.6 but then throws a runtime exception.  In my `install-pytorch.sh` script that gets run when you setup this project, it installs `pilow<9`, but sometimes it gets upgraded when the user installs other packages.

closed 2023-04-27 by dusty-nv