dusty-nv/jetson-inference #1535: jetson.utils -- compiled without NumPy array conversion support

olegzhos, 2022-12-18:
Hello everyone!
I have completed all the steps of Building the Project from Source (https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md) exactly and few times! But I have the next exception:
jetson.utils -- compiled without NumPy array conversion support (warning)
jetson.utils -- if you wish to have support for converting NumPy arrays,
jetson.utils -- first run 'sudo apt-get install python-numpy python3-numpy'
warning:  importing jetson.utils is deprecated.  please 'import jetson_utils' instead.

That means I can't import cudaToNumpy  and cudaFromNumpy from jetson.utils or jetson_utils too. I have python3-numpy installed because if I run python3 interpreter and the next commands in terminal I have:
python3
>>> import numpy as np
>>> print(np.__version__)
1.19.5

I was installing numpy (sudo apt-get install python-numpy python3-numpy) few times and few times re-building/re-installing project after that but it doesn't help in my case(

olegzhos, 2023-01-03:
@dusty-nv , could you reply to this question please? I'm sorry but I really have no idea how to avoid this problem

dusty-nv, 2023-04-27:
@olegzhos sorry for the delay if this is still an issue.  I would try `sudo apt-get --reinstall install python3-numpy python-numpy` or use the Docker container instead.

closed 2023-04-27 by dusty-nv