dusty-nv/jetson-inference #1538: cropping the frame in imagenet.py classification code

Farouq-Mot, 2022-12-30:
Hello, 

I trained my own classifier for traffic light color classification using this link:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md

Following your instructions, it runs on camera feed using this command:
imagenet.py --model=$NET/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt csi://0

Now I need to crop several traffic lights from each frame and run the classifier on them. How do I crop the traffic lights from each frame in the imagenet.py python code and run it?  Where is the imagenet.py code located? 

Please reply to this question since it is crucial for our application.

Regards,
Farouq

dusty-nv, 2022-12-30:
Do you know where the bounding-box coordinates of the traffic lights are that you want to crop?   If not, you'll need to use/train a detection model instead.  There is [`detectnet-snap.py`](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet-snap.py) which does a similar cropping thing with the detection results.

Farouq-Mot, 2022-12-30:
Yes I know  the traffic light coordinates.
My question is how to crop the frame in the imagenet.py code ?
Where is the imagenet.py code on the jetson? How di I change it?

dusty-nv, 2023-01-02:
The imagenet.py code is located in your jetson-inference directory under `jetson-inference/python/examples/imagenet.py`

After you make changes to this script, you can either run it from that directory, or re-run the following commands so that it is installed into your bin dirs:

``` bash
cd jetson-inference/build
cmake ../
make
sudo make install
```

There is some example of cropping an image with CUDA here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#cropping

Farouq-Mot, 2023-01-03:
Thanks so much ! great response.

Farouq-Mot, 2023-01-03:
Hi,  I tested the crop method. The crop works fine when I test it on a single image and the predicted class is also correct.
When I change the input from image to camera, the predicted class becomes wrong. What is the reason for this issue?

When using camera, In the out put that the imagenet.py prints on the console, it show only 2 classes (red, green) and their confidence but I have three classes (red, green, yellow).

Here is the output I get when using camera and the traffic light is yellow. Why it does not print the third class here?
class 0000 - 0.873937  (green)
class 0001 - 0.124984  (red)

**I printed the img and the cropped image info too:**

img <cudaImage object>
   -- ptr:      0x218161000
   -- size:     6220800
   -- width:    1920
   -- height:   1080
   -- channels: 3
   -- format:   rgb8
   -- timestamp: 116.789597
   -- mapped:   true
   -- freeOnDelete: false

crop_img <cudaImage object>
   -- ptr:      0x214059c00
   -- size:     4485
   -- width:    23
   -- height:   65
   -- channels: 3
   -- format:   rgb8
   -- timestamp: 116.789597
   -- mapped:   true
   -- freeOnDelete: true

crop_img width 23
crop_img height 65

**This is also the code  I added to imgenet.py:**

# capture the next image
	img = input.Capture()
	print('img',img)

	# allocate the output image, with the cropped size
	imgOutput = jetson.utils.cudaAllocMapped(width=right-left,height=bottom-top,format=img.format)

        # crop the image to the ROI
	jetson.utils.cudaCrop(img, imgOutput, crop_roi)

	# classify the image
	class_id, confidence = net.Classify(imgOutput)

	print('class_id', class_id)
	print('confidence', confidence)
	print('crop_img',imgOutput)
	print('crop_img width',imgOutput.width)
	print('crop_img height',imgOutput.height)

Please let me know how to fix this problem since my application is camera based.

Thanks so much for your help.

Farouq-Mot, 2023-01-04:
I found the issue. My light coordinates were not correct on the camera video feed.

closed 2023-04-27 by dusty-nv