dusty-nv/jetson-inference #1540: Can I resize of Bounding Box?

kookmin-DS-UAV, 2023-01-04:
Hello, 
Can I resize of bounding box to fit of object?
I'd like to know how to draw bounding box of jetson_inference.
(e.g. using utils's videoOption.cpp s/th)

Thanks
[image]

Marina

dusty-nv, 2023-01-04:
Hi @kookmin-DS-UAV, yes you can manually change the detection results returned from detectNet::Detect() and then use detectNet::Overlay() to draw them after you change them.   Also you'll want to disable detectNet::Detect() from doing the overlay by passing it `detectNet::OVERLAY_NONE` so that it doesn't do the overlay before you alter them.

kookmin-DS-UAV, 2023-01-05:
Hi @dusty-nv, thank you for your answer.
Would you let me know about detectNet::Detect() and detectNet::Overlay()?
I don't understand 'manually change the detection results returned from detectNet::Detect()' and also 'use detectNet::Overlay() to draw'.
Do I have to fixed cpp file?? (which one? and sorry, I'm not good at c++)

Thanks!

dusty-nv, 2023-01-05:
@kookmin-DS-UAV normally detectNet::Detect() can be called with overlay flags such that it automatically calls detectNet::Overlay() for you.   However if you call detectNet::Detect() with OVERLAY_NONE, it will skip doing the overlay.  Then you can do it yourself later after you modify the array of detection results.   You can also do this from Python too if you prefer.

closed 2023-04-27 by dusty-nv