dusty-nv/jetson-inference #1542: How to check the centroid of detected object.

Kant01354, 2023-01-05:
Dear sir,

How to print the centroid of the detected objects rectangular box ? Is there any function already for the same ?

dusty-nv, 2023-01-05:
Hi @Kant01354, you can use the detectNet.Detection.Center() function from C++ or Python to get the centroid:

https://github.com/dusty-nv/jetson-inference/blob/9ee9a950a80fbc7597d7e78a7ba0a282e85fae78/c/detectNet.h#L147

However I don't have it implemented to draw the centroid.  You could modify the detectNet::Overlay() function in C++ if you wanted to add that.

Kant01354, 2023-01-06:
Sir I tried this in detectnet.py code :-

```

	# print the detections
	print("detected {:d} objects in image".format(len(detections)))

	for detection in detections:
		print(detection)
		print (detectNet.Detection.Center() )

	# render the image
	output.Render(img)
```

But it crashed with the error -
_print (detectNet.Detection.Center() )
TypeError 'getset_descriptor' object is not callable_

dusty-nv, 2023-01-06:
Hi @Kant01354, do this instead:

``` python
for detection in detections:
     print(detection)
     print (detection.Center)
```

closed 2023-01-12 by Kant01354