dusty-nv/jetson-inference #1543: Docker build from source fails for 5.0.2

poett1, 2023-01-10:
Hello,

I'm trying to build the container from source (master branch) on Jetson Xavier AGX with Jetpack 5.0.2. Seems like there are some libraries missing for inference.
 
Code to reproduce this error:
```
git clone --recursive https://github.com/dusty-nv/jetson-inference
cd jetson-inference
docker/build.sh
```

Build fails with the following error:

```
....
[ 70%] Linking CXX shared library aarch64/lib/libjetson-inference.so
[ 70%] Built target jetson-inference
[ 72%] Building CXX object examples/imagenet/CMakeFiles/imagenet.dir/imagenet.cpp.o
[ 72%] Building CXX object examples/imagenet/CMakeFiles/imagenet-camera.dir/imagenet.cpp.o
[ 75%] Building CXX object examples/depthnet/CMakeFiles/depthnet.dir/depthnet.cpp.o
[ 75%] Building CXX object examples/imagenet/CMakeFiles/imagenet-console.dir/imagenet.cpp.o
[ 75%] Building CXX object examples/detectnet/CMakeFiles/detectnet.dir/detectnet.cpp.o
[ 76%] Building CXX object examples/detectnet/CMakeFiles/detectnet-console.dir/detectnet.cpp.o
[ 76%] Building CXX object examples/detectnet/CMakeFiles/detectnet-camera.dir/detectnet.cpp.o
[ 76%] Building CXX object examples/posenet/CMakeFiles/posenet.dir/posenet.cpp.o
[ 77%] Linking CXX executable ../../aarch64/bin/depthnet
[ 78%] Linking CXX executable ../../aarch64/bin/detectnet-console
[ 78%] Linking CXX executable ../../aarch64/bin/imagenet-camera
[ 79%] Linking CXX executable ../../aarch64/bin/posenet
[ 80%] Linking CXX executable ../../aarch64/bin/detectnet
[ 81%] Linking CXX executable ../../aarch64/bin/detectnet-camera
[ 82%] Linking CXX executable ../../aarch64/bin/imagenet
[ 83%] Linking CXX executable ../../aarch64/bin/imagenet-console
/usr/bin/ld: warning: libnvdla_compiler.so, needed by /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnvmedia.so, needed by /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnvmedia_tensor.so, needed by /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnvmedia_dla.so, needed by /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setUseSoftMaxOptz(bool)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaLoadableCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IPoolingLayer::setPoolingPaddingInclusionType(nvdla::PoolingPaddingInclusionType)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::destroyWisdom(nvdla::IWisdom*)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaGetMaxOutstandingTasks'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamConsumerDestroy'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaGetInputTensorDescriptor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamConsumerAcquireMetaData'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamProducerGetTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamProducerCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaDataUnregister'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::INetwork::addSlice(nvdla::ITensor*, nvdla::Weights, nvdla::Weights, nvdla::Weights, nvdla::Weights, nvdla::SliceLayerMode)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDeviceDestroy'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamProducerDestroy'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamConsumerCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaGetOutputTensorDescriptor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::destroyNetwork(nvdla::INetwork*)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::createWisdom()'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorLock'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaInit'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorGetMetaData'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::createNetwork()'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamConsumerAcquireTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaGetNumOfInputTensors'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaSubmit'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaRemoveLoadable'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaLoadLoadable'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamConsumerReleaseTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamProducerPostTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaAppendLoadable'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setCanCompressStructuredSparseWeights(bool)'
[... 366 lines cut ...]
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaLoadLoadable'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaCreate'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamConsumerReleaseTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaEglStreamProducerPostTensor'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaAppendLoadable'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `nvdla::IProfile::setCanCompressStructuredSparseWeights(bool)'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaTensorEglStreamProducerPostMetaData'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/9/../../../aarch64-linux-gnu/libnvinfer.so: undefined reference to `NvMediaDlaGetNumEngines'
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:487: examples/imagenet/CMakeFiles/imagenet-console.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c mkdir docs &&     touch docs/CMakeLists.txt &&     sed -i 's/nvcaffe_parser/nvparsers/g' CMakeLists.txt &&     mkdir build &&     cd build &&     cmake ../ &&     make -j$(nproc) &&     make install &&     /bin/bash -O extglob -c "cd /jetson-inference/build; rm -rf -v !(aarch64|download-models.*)" &&     rm -rf /var/lib/apt/lists/*' returned a non-zero code: 2
```

dusty-nv, 2023-01-10:
Hi @poett1, did you first set your default docker runtime to nvidia?   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

If you already did that, can you confirm that you do indeed have the libnvmedia drivers on your host device?

``` bash
ls -ll /usr/lib/aarch64-linux-gnu/tegra/libnvmedia*
-rw-r--r-- 1 root root 104520 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia_2d.so
-rw-r--r-- 1 root root  43080 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia2d.so
-rw-r--r-- 1 root root  34744 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia_dla.so
-rw-r--r-- 1 root root 125416 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia_isp_ext.so
-rw-r--r-- 1 root root  55400 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedialdc.so
-rw-r--r-- 1 root root 359752 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia.so
-rw-r--r-- 1 root root  68000 May 17  2022 /usr/lib/aarch64-linux-gnu/tegra/libnvmedia_tensor.so
```

poett1, 2023-01-11:
Setting default docker runtime to nvidia solved the problem, thanks!

closed 2023-01-11 by poett1