dusty-nv/jetson-inference #1544: Gstreamer pipeline automatically getting closed

krithicswaroopan, 2023-01-12:
I'm developing a jetbot project which performs its movements by detecting objects in front of it.
I use a custom **my-detection.py** file to perform the detections with instruction, where while running the file the gstreamer pipeline automatically close once the pipeline is opened, since the heavy load the openGL aslo fails to give a output stream. I have attached the files and the log below.

## My-detection.py
``` 
from jetson_inference import detectNet
from jetson_utils import videoSource, videoOutput
import time
import argparse
import sys  
from jetbot import Robot

net = detectNet(argv=["--model=/home/jetbot/jetson-inference/python/training/detection/ssd/models/tf3mobileNet_v1.onnx","--labels=/home/jetbot/jetson-inference/python/training/detection/ssd/models/labels.txt","--input-blob=input_0","--output-cvg=scores","--output-bbox=boxes"],threshold=0.5)
camera = videoSource("csi://0")      # '/dev/video0' for V4L2
display = videoOutput("display://0") # 'my_video.mp4' for file

while True:
	img = camera.Capture()
	detections = net.Detect(img)
	# print the detections
	print("detected {:d} objects in image".format(len(detections)))

	for detection in detections:
		print(detection)
		if opt.use_car:
			start_car(detection)

	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
	
	# print out performance info
	net.PrintProfilerTimes()

	if KeyboardInterrupt:
		#not camera.IsStreaming() or not display.IsStreaming() 
		#jetson_utils.gstCamera.Close()  
		break

def start_car(detection):
	robot.stop()
	if detection.ClassID == 0 or detection.ClassID == 11 :   #BACKGROUND & signal green
		robot.forward(0.2)
	elif detection.ClassID == 1 or detection.ClassID == 12 :   #cross walk & Road Cross walk
		robot.stop()
		time.sleep(3)
		robot.forward(0.2)
	elif detection.ClassID == 2 :   #gate closed
		robot.stop()
	elif detection.ClassID == 3 :   #gate opened
		robot.forward(0.2)

	elif detection.ClassID == 4 :   #stop
		robot.stop()
		time.sleep(20)
		robot.forward(0.2)
	elif detection.ClassID == 5 :   #no entry
		robot.stop()
		robot.right(0.1)
		time.sleep(2)
		robot.forward(0.2)
	elif detection.ClassID == 6 :   #speed limit 50	
		robot.forward(0.5)
		time.sleep(10)
		robot.forward(0.2)
	elif detection.ClassID == 7 or detection.ClassID == 8 or detection.ClassID == 9 or detection.ClassID == 10 or detection.ClassID == 11 :
		robot.forward(0.1)
	elif detection.ClassID == 12 or detection.ClassID == 13 or detection.ClassID == 14 or detection.ClassID == 15 or detection.ClassID == 16 :
		robot.forward(0.1)
```
## Terminal Log
[... 185 lines cut ...]
detected 0 objects in image

[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetbot/jetson-inference/python/training/detection/ssd/models/tf3mobileNet_v1.onnx
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.08771ms  CUDA   3.18260ms
[TRT]    Network       CPU 493.60477ms  CUDA 476.01093ms
[TRT]    Post-Process  CPU   6.45864ms  CUDA   6.68385ms
[TRT]    Total         CPU 500.15112ms  CUDA 485.87741ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
[gstreamer] gstCamera -- pipeline stopped
```
I'm working on the solution to let the program run and the pipeline to be closed when a keyboard Interrupt is observed.
or procedure to declare the **mStreaming  = False** from the my-detection.py file

looking for Insights and solution for the scenario.
@dusty-nv

dusty-nv, 2023-01-12:
```
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
```

Hi @krithicswaroopan, that message only gets printed out when the stream is closed/destroyed by the user:   https://github.com/dusty-nv/jetson-utils/blob/8b6c5ca4b2b52d51f5415ce033ee52563ddf3f2c/camera/gstCamera.cpp#L750

Are you sure your interrupt isn't being triggered prematurely?

Also I think you want to have your `start_car()` function run in a different thread, because it has calls to `sleep()` in it which will block the camera processing pipeline.

krithicswaroopan, 2023-01-17:
Hi @dusty-nv , thankyou for you response.
 yes, as you referenced that means the pipeline is closed it's obvious 
But in my case, I wanted to close the pipeline manually with some trigger as like ``KeyboardInterrupts``, have checked and sure that the interrupt is not triggered prematurely.
**In this case can you suggest me a solution to close the pipeline manually once the detection starts ?**

Even when all the ``sleep()`` functions are removed from the code the same problem occurs, where the pipeline gets closed immediately after initialization.

dusty-nv, 2023-01-17:
@krithicswaroopan if you run `video-viewer.py csi://0` or `detectnet.py csi://0` does the camera run correctly?

If so, I think your code may be closing the stream.  Where is `KeyboardInterrupt` defined?

krithicswaroopan, 2023-01-24:
@dusty-nv Thank you for your insights, I have solved the issue.
The problem was the improper definition of ``KeyboardInerrupt``. I have the pipeline is being closed properly

closed 2023-01-24 by krithicswaroopan