dusty-nv/jetson-inference #1545: Color detection does not work for mobilessd and detectnet, followed guide from S3E5

gd08xxx, 2023-01-14:
I followed the custom object detection example from S3E5 and tried to differentiate the color of different m&ms on a table with bounding boxes. 

I trained the model with the camera-capture tool and stated the areas of different colored m&ms on a table. It turns out there are no bounding boxes created for the color of each m&m when I run detectnet.

However, when I just train the model with only m&m in labels.txt, ignoring the color, detectnet creates bounding boxes for each m&m, saying which area contains an m&m without stating its color.

The below labels.txt does not work
‘’’
red
green
orange
yellow
blue
‘’’

The below labels.txt works
‘’’
m&m
‘’’
I used the mobilessd framework to train the model from the video example

gd08xxx, 2023-01-16:
@dusty-nv any chance you can give me a direction to solve this problem? Like changing the library, do more training? (I have trained 50 images of random number of m&ms) From my understanding, after training in pytorch, there will be a colour filter applied to the images.

Help will really be appreciated
Thanks,
Godfrey

dusty-nv, 2023-01-17:
@gd08xxx when you created the dataset with the camera-capture tool, did you label the bounding boxes with the colors, or just m&m?   It seems odd that the m&m labels.txt works because the annotated data would need to match that string, which leads me to believe the answer is the later.

Can you post one of the XML files from your dataset?   Also 50 images is probably not enough, but let's start with the classes first.

closed 2023-04-27 by dusty-nv

gd08xxx, 2023-05-07:
I found out that in python/training/detection/ssd/vision/data_preprocessing.py, there is a line of code named PhotometricDistorct() in TrainAugmentation's init
```
def __init__(self, size, mean=0, std=1.0):
        """
        Args:
            size: the size the of final image.
            mean: mean pixel value per channel.
        """
        self.mean = mean
        self.size = size
        self.augment = Compose([
            ConvertFromInts(),
            # MODIFIED
            # PhotometricDistort(),
            Expand(self.mean),
            RandomSampleCrop(),
            RandomMirror(),
            ToPercentCoords(),
            Resize(self.size),
            SubtractMeans(self.mean),
            lambda img, boxes=None, labels=None: (img / std, boxes, labels),
            ToTensor(),
        ])
```
After commenting it out, color detection works, this is because the PhotometricDistort augments the colors the m&ms before training

dusty-nv, 2023-05-08:
Thanks for sharing that @gd08xxx, during data augmentation some degree of color jitter is normally beneficial for being more robust against different camera/lighting conditions, but this is good to remember when detecting/classifying objects based on their color...