dusty-nv/jetson-inference #1546: Custom data classification model Hardly able to classify anything even after training the model after following S3E3 Training Image Classification Models.

SaTz1483, 2023-01-15:
Hey ! @dusty-nv Could you please help me out here !

I have followed the instructions exactly as told in S3E3, https://www.youtube.com/watch?v=sN6aT9TpltU video for training this classification resne18 model on a custom dataset, before trying my custom dataset I re-trained the cat-dog to check if everything is fine and it worked fine.

but now I have trained the model with a custom dataset consisting of 26 classes and around 24000 images, for 35 epochs and 16 batch sizes using the python3 train.py file in the repo. however even after training if i feed the model with an image from the test folder or run the stream from my csi port. it can hardly detect anything and accuracy is stuck at 8.05 % no matter what I show in the camera-stream or input as a image.

i'd also like to mention the jetson nano lost power mid-training at around 27 epochs, and i resumed the training using `--resume`
argument and it continued and finished all 35 epochs then now has saved the model to checkpoint.pth. but when I run the imagenet command does it use the modelbest.pth or checkpoint.pth.? I'm still suffering with the same issue. 

I also studied the graph given in the repo it states that 30 epochs for a dataset of 5000 images gives the best accuracy. does that mean I have to change any hyper-parameters? 

my directory management is also exactly as stated in this repo instructions. and i haven't used the docker container i run it locally on my jetson nano itself.

dusty-nv, 2023-01-15:
Hi @SaTz1483, can you try deleting the *.engine file from your model's directory and try running again?

Also what was the best accuracy in pytorch when you trained the model?

SaTz1483, 2023-01-16:
hey @dusty-nv 

i have deleted the engine file and run the detection i still am facing the same issue.

these are the logs i received when i ran the train.py file with -e argument (evaluate).

python3 train.py data/SUAS_LEVEL -e

Use GPU: 0 for training
=> dataset classes: 26 [‘A’, ‘B’, ‘C’, ‘D’, ‘E’, ‘F’, ‘G’, ‘H’, ‘I’, ‘J’, ‘K’, ‘L’, ‘M’, ‘N’, ‘O’, ‘P’, ‘Q’, ‘R’, ‘S’, ‘T’, ‘U’, ‘V’, ‘W’, ‘X’, ‘Y’, ‘Z’]
=> using pre-trained model ‘resnet18’
=> reshaped ResNet fully-connected layer with: Linear(in_features=512, out_features=26, bias=True)
Test: [ 0/212] Time 34.410 (34.410) Loss 2.5497e+00 (2.5497e+00) Acc@1 25.00 ( 25.00) Acc@5 87.50 ( 87.50)
Test: [ 10/212] Time 0.203 ( 3.728) Loss 2.6695e+00 (2.5342e+00) Acc@1 37.50 ( 48.86) Acc@5 87.50 ( 92.05)
Test: [ 20/212] Time 0.214 ( 2.075) Loss 3.6322e+00 (3.1246e+00) Acc@1 0.00 ( 25.60) Acc@5 0.00 ( 48.81)
Test: [ 30/212] Time 0.206 ( 1.471) Loss 3.0034e+00 (3.1224e+00) Acc@1 0.00 ( 17.34) Acc@5 25.00 ( 41.13)
Test: [ 40/212] Time 0.197 ( 1.161) Loss 3.9464e+00 (3.2332e+00) Acc@1 0.00 ( 13.11) Acc@5 0.00 ( 33.23)
Test: [ 50/212] Time 0.198 ( 0.972) Loss 4.7079e+00 (3.3412e+00) Acc@1 0.00 ( 10.54) Acc@5 0.00 ( 26.72)
Test: [ 60/212] Time 0.197 ( 0.845) Loss 3.4375e+00 (3.3977e+00) Acc@1 0.00 ( 8.81) Acc@5 0.00 ( 22.54)
Test: [ 70/212] Time 0.199 ( 0.754) Loss 3.3784e+00 (3.3897e+00) Acc@1 0.00 ( 7.57) Acc@5 0.00 ( 19.89)
Test: [ 80/212] Time 0.198 ( 0.685) Loss 3.8194e+00 (3.4061e+00) Acc@1 0.00 ( 6.64) Acc@5 0.00 ( 18.21)
Test: [ 90/212] Time 0.196 ( 0.632) Loss 2.7763e+00 (3.4055e+00) Acc@1 0.00 ( 6.59) Acc@5 87.50 ( 19.37)
Test: [100/212] Time 0.196 ( 0.589) Loss 3.5806e+00 (3.3505e+00) Acc@1 0.00 ( 7.18) Acc@5 0.00 ( 24.50)
Test: [110/212] Time 0.199 ( 0.554) Loss 2.7290e+00 (3.3543e+00) Acc@1 25.00 ( 6.76) Acc@5 75.00 ( 23.09)
Test: [120/212] Time 0.197 ( 0.524) Loss 2.8261e+00 (3.3200e+00) Acc@1 0.00 ( 6.51) Acc@5 75.00 ( 25.83)
Test: [130/212] Time 0.198 ( 0.499) Loss 2.5719e+00 (3.2694e+00) Acc@1 37.50 ( 8.30) Acc@5 87.50 ( 30.63)
Test: [140/212] Time 0.199 ( 0.478) Loss 3.6897e+00 (3.2806e+00) Acc@1 0.00 ( 7.89) Acc@5 0.00 ( 29.61)
Test: [150/212] Time 0.197 ( 0.459) Loss 3.3786e+00 (3.2937e+00) Acc@1 0.00 ( 7.37) Acc@5 12.50 ( 28.15)
Test: [160/212] Time 0.197 ( 0.443) Loss 3.3722e+00 (3.3019e+00) Acc@1 0.00 ( 6.91) Acc@5 12.50 ( 26.86)
Test: [170/212] Time 0.198 ( 0.429) Loss 3.1783e+00 (3.3070e+00) Acc@1 0.00 ( 6.51) Acc@5 25.00 ( 25.58)
Test: [180/212] Time 0.196 ( 0.416) Loss 3.0151e+00 (3.3000e+00) Acc@1 0.00 ( 6.15) Acc@5 25.00 ( 25.21)
Test: [190/212] Time 0.196 ( 0.404) Loss 3.4193e+00 (3.2844e+00) Acc@1 0.00 ( 5.96) Acc@5 12.50 ( 26.77)
Test: [200/212] Time 0.197 ( 0.394) Loss 3.3029e+00 (3.2888e+00) Acc@1 0.00 ( 5.66) Acc@5 12.50 ( 25.75)
Test: [210/212] Time 0.194 ( 0.385) Loss 3.3156e+00 (3.2926e+00) Acc@1 0.00 ( 5.39) Acc@5 0.00 ( 24.94)

Acc@1 5.378 Acc@5 24.882

dusty-nv, 2023-01-17:
@SaTz1483 the accuracy is only 5%.   What is the accuracy when it is done training?

SaTz1483, 2023-01-17:
@dusty-nv 

I had the same accuracy even when it was done training, infact the accuracy values remained the same even after every epoch here js a pic i had taken during the 6th epoch. [image]

dusty-nv, 2023-01-17:
Typically that means the model is not converging and you can try changing the learning rate, or perhaps the dataset is too difficult for the model.  I'm not an expert on training and this code comes from the PyTorch examples.  

What resolution are the images?   Are they in color?

SaTz1483, 2023-01-19:
@dusty-nv  they are 100x100 images in the color of objects, that have a letter and background of different colors,
[image]

SaTz1483, 2023-01-19:
is there any way around this maybe using the detection algorithm or anything that I could train these images?
what would be the ideal number of images I could train it with so the model converges?

dusty-nv, 2023-01-19:
@SaTz1483 the standard resolution for ResNet is 224x224, so it might be that these images are too small for the network to effectively use.  I'm not an expert on training though, so it may require some experimentation from you to resolve.

SaTz1483, 2023-01-19:
@dusty-nv 
Thanks for your help ! I'll do some experimentation with number of images and the resolution of that data I'm using and let you know.

SaTz1483, 2023-01-25:
@dusty-nv  is this an issue I'm getting  soctherm: OC ALARM 0x0000001
to note that i wrote a python code to generate the XML files exactly same as the given examples in openimages 

I also trained on it with different data before i dint get that line before while training. Does this mess with my training accuracy or is this error even related to training?

[image]

SaTz1483, 2023-01-25:
@dusty-nv  is this an issue I'm getting  soctherm: OC ALARM 0x0000001
to note that i wrote a python code to generate the XML files exactly same as the given examples in openimages 

I also trained on it with different data before i dint get that line before while training. Does this mess with my training accuracy or is this error even related to training?

Now it comes if i train any other model as well.

[image]

dusty-nv, 2023-01-25:
Hi @SaTz1483, that error is unrelated to training - you may want to try putting a fan on your Nano or cooling it down.

I haven't done custom OpenImages datasets, only custom Pascal VOC datasets.  Pascal VOC is a simpler format to work with IMO

dusty-nv, 2023-01-25:
OK cool, yes Pascal VOC is good.  It seems that originally you were training a classification model and now you are trying detection model.  It may be this is just a challenging dataset to train.  If you try running this in Google Colab or another machine with PyTorch+GPU, it may help you train/experiment faster.  You can run my training scripts on other devices that have GPU.

SaTz1483, 2023-01-25:
@dusty-nv 

May i please know where I can find the colab notebook for training? For both classification and detection

dusty-nv, 2023-01-25:
I don't have notebooks for it, you would need to write a notebook which launched the python script

SaTz1483, 2023-01-25:
@dusty-nv 

Thanks for your help!

dusty-nv, 2023-01-25:
No problem @SaTz1483, hope you are able to get it working!

SaTz1483, 2023-01-29:
> @SaTz1483 the standard resolution for ResNet is 224x224, so it might be that these images are too small for the network to effectively use. I'm not an expert on training though, so it may require some experimentation from you to resolve.

@dusty-nv here when you mention the standard resolution is 224x224 do you mean the images in the dataset must all be this resolution ? does the model take care of the resizing ? or should we resize/crop them? beacuse now my images are smaller than this standard.

SaTz1483, 2023-02-04:
@dusty-nv 

I have managed to get everything working as of now, i have one doubt regarding hyperparameters.

I saw a graph on this repository that said when SSD-mobilenet was trained for 5000 images at 30 epochs it gave best accuracy.

Right now i have a dataset of 11000 images and another of 23000 images 
Should I train with the same hyperparameters ? Or make any changes like increasing the epochs or batch-size?
[image]

dusty-nv, 2023-02-05:
Hi @SaTz1483, you could train for more epochs yes (like 50 or 100).  The updated code logs to tensorboard so you could keep an eye on the training progress easier.  Since your dataset is larger you could run github.com/dusty-nv/pytorch-ssd on x86+GPU system if you have access to one (or cloud).  I typically train with the largest batch size that will fit into memory.  I haven't messed around with other hyperparamers like learning rate or LR scheduling.

SaTz1483, 2023-02-05:
@dusty-nv  oh okay ,

SaTz1483, 2023-02-05:
@dusty-nv  

How do I resume my training on detection algorithm?

 Beacuse there is no checkpoint.pth in models directory. When i try to use the --resume argument i can't really find the checkpoint.pth file, this was there only for the classification model.

dusty-nv, 2023-02-06:
> Beacuse there is no checkpoint.pth in models directory. When i try to use the --resume argument i can't really find the checkpoint.pth file, this was there only for the classification model.

Hi @SaTz1483, at the end of each epoch, a checkpoint is saved to your model's directory like `mb1-ssd-Epoch-0-Loss-2.662300308942795.pth`

So you would pick from the last epoch trained. I would probably use a new model directory when you resume.  Then if it starts from 0 it's clear which checkpoints are from which run.

> 1. does this mean i can run this docker container on an nvidia DGX system with no problems? Or build project from source?(i have access to one of these)

I haven't tried it on DGX, just my local PC with a GeForce card in it.  Some other people have tried it on their machines ok.

> 2\. How do you know that a certain value of your batch-size is the highest that will fit into your memory? Trial and error?

Yea by working up and by watching the system memory / GPU memory to judge when it's getting close to running out.  Within reason.

> 3\. what all requirements would i need to have?
>     just run the train_ssd.py on cloud with the data folder or should also i have the models and everything built from source on the cloud?

Offhand...PyTorch, torchvision, pandas, and what's in pytorch-ssd/requirements.txt

SaTz1483, 2023-02-18:
Hey @dusty-nv 

When i try to build the dockerfile it throws me this error I'm trying to install these dependencies. And this is how I've changed the dockerfile 

apt-get install inotify-tools
[image]

But it throws me this error 

[image]