dusty-nv/jetson-inference #1547: Loss of performance after retraining

MarcFacerias, 2023-01-16:
Hello, I retrained the ssd-mobilenet following the scripts you provided and when exporting it to onnx and using the detectnet.py with it I saw a huge increase of inference speed. I'm using the Jetson Nano and trained and exported the network in another device, do you have any clue of what could be happening? 

Thanks a lot

dusty-nv, 2023-01-17:
Hi @MarcFacerias, what model are you judging the increase in inference speed against?    Or do you mean that the same model on two different Jetson Nano's runs differently?

MarcFacerias, 2023-01-22:
I'm sorry for the late reply, I'm retraining the ssd mobilenet with a custom dataset in google colab using your train script and the inference time is way longer than with the coco dataset ( off the shelf version that you provided )

dusty-nv, 2023-01-23:
How many classes are in your custom dataset?   Are you deploying it with detectNet?

The COCO model is a TensorFlow / UFF model, so it behaves differently.

MarcFacerias, 2023-01-23:
I have 4 classes  and I'm deploying with detectnet.py

dusty-nv, 2023-01-23:
Hmm okay, in my experience the PyTorch-trained ssd-mobilenet should be pretty fast when it is using smaller number of classes.  What is the difference in performance of your ONNX model versus the COCO model?

BTW one thing I've been meaning to try is the ONNX simplifier tool:   https://github.com/daquexian/onnx-simplifier

MarcFacerias, 2023-01-24:
I can't check the numbers right now but off the top of my head the inference time of the retrained model was around 1 sample per second, later this week I'll reach you back with the actual comparison and I'll try this tool. Thanks !

MarcFacerias, 2023-01-30:
Sorry for the delay, I did some tests and whenever I make the inference over a batch of inputs the inference time is the same as in your tests, however when I try to evaluate a single sample it increases up to few seconds. Could you help me understand why this could be happening ? Thanks a lot, I appreciate your help

dusty-nv, 2023-01-30:
Have you tried running `jetson_clocks` script beforehand?   This will disable dynamic clock frequency scaling.

When you just process a single image, the GPU doesn't have time to ramp up to it's maximum frequency.  For actual performance analysis, it's recommended to process a video stream.

MarcFacerias, 2023-01-31:
I see, thanks a lot, as you mentioned just looping the inference code matches the performance presented in other tests.

closed 2023-01-31 by MarcFacerias