dusty-nv/jetson-inference #1551: What the "Timiong Report" mean?

k-nxt, 2023-01-26:
I've run the code 
`./imagenet.py images/jellyfish.jpg /jetson_inference/jellyfish.jpg`  
in https://pytorch.org/blog/running-pytorch-models-on-jetson-nano/

and got the log of Timing Report

```
[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/bvlc_googlenet.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.05328ms  CUDA   0.03840ms
[TRT]    Network       CPU   4.47820ms  CUDA   4.85386ms
[TRT]    Post-Process  CPU   0.04013ms  CUDA   0.00147ms
[TRT]    Total         CPU   4.57161ms  CUDA   4.89373ms
[TRT]    ------------------------------------------------
```

What does this mean?

is this a comparison of the time between when we run all the process only "CPU" and "GPU(CUDA)"?
I'm confused because CPU time is faster than CUDA one ifor network.

Environment:
- Device : Jetson AGX Orin
- cat /etc/nv_tegra_release
># R34 (release), REVISION: 1.1, GCID: 30414990, BOARD: t186ref, EABI: aarch64, DATE: Tue May 17 04:20:55 UTC 2022

Thank you in advance/

dusty-nv, 2023-01-26:
Hi @k-nxt, CPU time is the wall clock time that elapses while the GPU is running.  Typically, the GPU time would be shorter than the CPU time (especially if the CPU is doing extra post-processing that the GPU doesn't do).   However they are using different kind of timers so there may be some minor variations at the microsecond/nanosecond level.  In this case, the milliseconds are the same and that makes sense.

k-nxt, 2023-01-27:
Hi @dusty-nv, thank you for your answer.

so in the case of this:
>[TRT]    Pre-Process   CPU   0.05328ms  CUDA   0.03840ms

Is my understanding correct that the total processing time is 0.03840ms (slower time)?

also, in the case of the result was this:
>[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/bvlc_googlenet.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   **0.05328ms**  CUDA   0.03840ms
[TRT]    Network       CPU   4.47820ms  CUDA   **4.85386ms**
[TRT]    Post-Process  CPU   **0.04013ms**  CUDA   0.00147ms
[TRT]    Total         CPU   4.57161ms  CUDA   **4.89373ms**
[TRT]    ------------------------------------------------

the total processing time = 0.05328ms(CPU) + 4.85386ms(CUDA) +  0.04013ms(CPU) + 4.89373ms(CUDA)
I chose the slower time for each part.

Is this correct?

dusty-nv, 2023-01-27:
@k-nxt you don't need to add the times together to get the total time, the CPU time encompasses both.  You can just go with the CPU time.  It's an aberration/inaccuracy in the timers that the CPU time is sometimes shorter than the CUDA time.

k-nxt, 2023-01-30:
@dusty-nv I understood the way to calculate the total time.
Thank you for your help!

closed 2023-01-30 by k-nxt