dusty-nv/jetson-inference #1553: Larger SSD model retrain

thanhngotud, 2023-01-31:
Hi,

I see you "added support for variable model resolutions". In this case, when we train the larger model, we just need to change the resolution right? Such as: python3 train_ssd.py --data=data/Error --model-dir=models/Error300 --batch-size=4 --epochs=30  --dataset-type=voc --resolution=640.

But I got a problem with training the model at 640x640. The loss is very high. It is always around 4.6. I keep training for 150 epochs but the loss does not decrease anymore.

sezer-muhammed, 2023-02-19:
I also tried this to train a model. How can we decide input res from jetson inference? I'm not sure if I train model at res 640, jetson inference will use it at correct res.
I guess high loss is normal since resolution it doubled and there is no normalizer so loss is higher I guess

dusty-nv, 2023-02-19:
> I'm not sure if I train model at res 640, jetson inference will use it at correct res.

Yes, jetson-inference automatically determines the resolution that the model uses from it's input layer.   The incoming camera stream then gets downsampled to this resolution.   

I can't comment on the higher loss, but that explanation sounds plausible @sezer-muhammed.  You could also run train_ssd.py with `--validate-mean-ap` to generate per-class Mean Average Precision (mAP) statistics during the validation pass at the end of each training epoch.  Hopefully that gives you a metric that's more quantifiable than just the raw losses.

closed 2023-04-27 by dusty-nv