dusty-nv/jetson-inference #1555: Custom model ERROR on segmentation

jsshin98, 2023-02-05:
Hi, Dusty.

I retrained fcn_resnet18 using your pytorch-segmentation project page.
I successfully exported onnx file and tried to run the model on live streaming.
However, I am getting this error : cannot find binding of given name : data.

How can I fix this error?:(  I already checked that it worked fine on your provided pretrained onnx file.

I simply type `./segnet.py --network=<MYNETWORKPATH> /dev/video0`
Here is my log

`[TRT]    device GPU, loaded /jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18_base.onnx
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 376, GPU 7871 (MiB)
[TRT]    Loaded engine size: 23 MiB
[TRT]    Deserialization required 10760 microseconds.
[TRT]    [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +22, now: CPU 0, GPU 22 (MiB)
[TRT]    Total per-runner device persistent memory is 15872
[TRT]    Total per-runner host persistent memory is 50496
[TRT]    Allocated activation device memory of size 4915200
[TRT]    [MemUsageChange] TensorRT-managed allocation in IExecutionContext creation: CPU +0, GPU +5, now: CPU 0, GPU 27 (MiB)
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       25
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 4915200
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  3
                -- dim #2  320
                -- dim #3  320
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  21
                -- dim #2  10
                -- dim #3  10
[TRT]    
[TRT]    3: Cannot find binding of given name: data
[TRT]    failed to find requested input layer data in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to create TensorRT engine for /jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18_base.onnx, device GPU
[TRT]    segNet -- failed to load.
jetson.inference -- segNet failed to load network
Traceback (most recent call last):
  File "./segnet.py", line 56, in <module>
    net = jetson.inference.segNet(opt.network, sys.argv)
Exception: jetson.inference -- segNet failed to load network

jsshin98, 2023-02-05:
I also put some arguments following other issues.
./segnet.py --network=/jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18_base.onnx --labels=/jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/classes.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video0

Then, I got other errors like this.
[TRT]    device GPU, loaded /jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18_base.onnx
[TRT]    [MemUsageChange] Init CUDA: CPU +0, GPU +0, now: CPU 376, GPU 8010 (MiB)
[TRT]    Loaded engine size: 23 MiB
[TRT]    Deserialization required 20075 microseconds.
[TRT]    [MemUsageChange] TensorRT-managed allocation in engine deserialization: CPU +0, GPU +22, now: CPU 0, GPU 22 (MiB)
[TRT]    Total per-runner device persistent memory is 15872
[TRT]    Total per-runner host persistent memory is 50496
[TRT]    Allocated activation device memory of size 4915200
[TRT]    [MemUsageChange] TensorRT-managed allocation in IExecutionContext creation: CPU +0, GPU +5, now: CPU 0, GPU 27 (MiB)
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       25
[TRT]       -- maxBatchSize 1
[TRT]       -- deviceMemory 4915200
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  3
                -- dim #2  320
                -- dim #3  320
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  4
                -- dim #0  1
                -- dim #1  21
                -- dim #2  10
                -- dim #3  10
[TRT]    
[TRT]    binding to input 0 input_0  binding index:  0
[TRT]    binding to input 0 input_0  dims (b=1 c=3 h=320 w=320) size=1228800
[TRT]    3: Cannot find binding of given name: score_fr_21classes
[TRT]    failed to find requested output layer score_fr_21classes in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to create TensorRT engine for /jetson-inference/data/networks/FCN-ResNet18-Pascal-VOC-320x320/fcn_resnet18_base.onnx, device GPU
[TRT]    segNet -- failed to load.
jetson.inference -- segNet failed to load network

jsshin98, 2023-02-05:
Oh I solved it by changing my command. Thanks for your wonderful code!

closed 2023-02-05 by jsshin98

dusty-nv, 2023-02-06:
Hi @jsshin98, thanks, yes I was going to recommend using `--output-blob=output_0` instead.