dusty-nv/jetson-inference #1556: jetson-inference on jetson agx orin L4T 20.04,  jetpack 5.1

malcolmcapel, 2023-02-05:
Greetings

Trying to get jetson-inference docker container working on a jetson agx orin runing ubuntun 20.04 and jetpack 5.1.  I flashed the L4T system and jetpack on an NVME drive (not the emm drive).  On boot root is associated with the NVME drive.  I've tried pulling the most recent jetson-inference container from your docker site and cloned the jetson-inference distro from github.  When I run docker/run.sh from ~/jetson-inference I get an indication that there is no local installation of jetsion-inference 3.5.1 found.

Thanks much

dusty-nv, 2023-02-06:
Hi @malcolmcapel, the container images for JetPack 5.1 are in-progress, for now you still use:  `docker/run.sh -c dustynv/jetson-inference:r35.1.0`

malcolmcapel, 2023-02-06:
are there any issues with having the root file system installed on an NVME rather than EMMC memory?

dusty-nv, 2023-02-06:
@malcolmcapel I haven't put the rootfs on NVME before, but I don't think so - I routinely put the docker root and my git repos on NVME

malcolmcapel, 2023-02-06:
Dusty...many thanks for the help!

I just tried to install R35.1 via the method you suggested.
After the installation completed I get:

....
V4L2_DEVICES:
localuser:root being added to access control list
Unable to find image 'dusty/jetson-inference:r35.10' locally 
docker: Error respone from daemon: mainfest fo rdustynv/jetson-inference:r35.10 not found: manifest unknown: manifest:unknown
See 'codker run --help'

thanks,
M

malcolmcapel, 2023-02-06:
I stupidly presumed that the run.sh script would pull the correct docker container.  I manually pulled r35.0.1 and now run.sh succeeded in starting the container.  Thanks again.

M

dusty-nv, 2023-02-06:
@malcolmcapel okay great - run.sh typically does pull the correct container by querying your JetPack version, but the jetson-inference container for JetPack 5.1 hasn't been released yet, so it's not working.  Sorry about that - glad you got it working!

malcolmcapel, 2023-02-06:
Dustin

I’ve exercised a bunch of stuff in the jetson-inference container…including retraining SSD-mobilenet with the fruit image subset (all 6000 or so) and 150 epochs, bastch-size of 4 and workers=4.  The retraining completed in about 4.5 hours.  I was able to invoke detectnet using the test fuit image set and got great statisitical detection results (as expected with this many epochs).

However, anything inside the jetson-inference containter  that uses gstreamer for video camera access and control fails.  I have a HD Pro webcam C920 attached (logitech) and  v4l2-ctl –list-devices shows:

              HD Pro Webcam C920 (usb-3610000.xhci-4.2):
                             /dev/video0
                             /dev/video1
                             /dev/media1

Along with some devices associated with NV Tegra input.

Anyway,
              Detectnet –model=models/fruit/ssd-mobilenet.onnx –labels=models/fruit/labels.txt \
                             --input-blob=input_0 –output-cvg=score –output-bbox=boxes \
                             /dev/video

Results in a bunch of errors indicating the nonfunctionality of the gstreamer handling of the webcam.
at the end of processing the test images.

In contrast, outside of the jetson-inference container gstreamer seem to be functional:

              nvgstcapture-1.0 –camsrc=0  results in a functioning gstreamer video stream from the webcam in a window.

Any idea about how to repair-reinstall-etc gstreamer components within the context of the jetson-inference docker container?

dusty-nv, 2023-02-06:
Can you post the error log here please?

Also, until the updated container is released, you could build jetson-inference from source and run it outside of docker.

malcolmcapel, 2023-02-06:
Dustin

Do you mean the dmesg output or does the container run an error log?

Thanks
M

dusty-nv, 2023-02-07:
Hi @malcolmcapel, I mean the terminal output from when you run the camera with detectnet or video-viewer program, to show what is going wrong with the camera

malcolmcapel, 2023-02-07:
Dustin..sorry for the delay

When I run: video-viewer /dev/video0 (within the jetson-inference container) I get the following output:
The output is obviously truncated….it ends with a looping “[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame”

looks like the module “nvidia” isn’t loaded.

Thanks,
Malcolm

modprobe: FATAL: Module nvidia not found in directory /lib/modules/5.10.104-tegra
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: HD Pro Webcam C920
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"HD\ Pro\ Webcam\ C920", v4l2.device.bus_info=(string)usb-3610000.xhci-4.2, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 53 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1536, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1296, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/2, 5/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 10/1, 15/2, 5/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [11] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [12] video/x-raw, format=(string)YUY2, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [13] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [14] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [15] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [16] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [17] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [18] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)90, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [19] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [20] image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [21] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [22] image/jpeg, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [23] image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [24] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [25] image/jpeg, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
^Creceived SIGINT
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
video-viewer:  shutdown complete
***@***.***:/jetson-inference#

dusty-nv, 2023-02-08:
> modprobe: FATAL: Module nvidia not found in directory /lib/modules/5.10.104-tegra

OK yes, it appears to be an issue from running the older container, sorry about that.  I hope to have the updated container soon.  In the meantime you could run it outside of container (sorry about that)

whab, 2023-02-08:
> I stupidly presumed that the run.sh script would pull the correct docker container. I manually pulled r35.0.1 and now run.sh succeeded in starting the container. Thanks again.
> 
> M

Hello, may I ask how you manually pull the correct docker container? I tried cloning the L4T-R35.1.0 branch, but I still have the same issue you described. Is it just me or is it that cumbersome to get the Jetson AGX Orin Kit working on some basic demos??? Thank you!

whab, 2023-02-08:
> Hello, may I ask how you manually pull the correct docker container? I tried cloning the L4T-R35.1.0 branch, but I still have the same issue you described. Is it just me or is it that cumbersome to get the Jetson AGX Orin Kit working on some basic demos??? Thank you!

Ok, found out on my own: sudo docker pull dustynv/jetson-inference:r35.1.0

dusty-nv, 2023-02-10:
OK, the jetson-inference container for JetPack 5.1 / R35.2.1 is now uploaded at `dustynv/jetson-inference:r35.2.1`

You should just now be able to run `docker/run.sh` on JetPack 5.1 and it should pull the correct container 👍

malcolmcapel, 2023-02-10:
Thanks!
Can you provide a procedure for upgrading to R35.2.1 from R35.1.0 that doesn't disturb the data contained on ./jetson-inference at the time of upgrade?

dusty-nv, 2023-02-10:
It won't delete any data unless you do docker/build.sh

malcolmcapel, 2023-02-10:
sorry to belabor this but should we do a docker ri on the old container to recover disk space?

dusty-nv, 2023-02-10:
Sure yea, you could do that

malcolmcapel, 2023-02-10:
confirmed...a simple docker/run.sh pulled R35.2.1 under jetpack 5.1 on my orin....thanks much for the help!

malcolmcapel, 2023-02-11:
Dustin:

Sorry to report still getting the NVIDIA module not loaded indicated by modprobe under R35.2.1 on my orin:

root@ubuntu:/jetson-inference/build# cd aarch64/bin
root@ubuntu:/jetson-inference/build/aarch64/bin# ./video-viewer /dev/video0
modprobe: FATAL: Module nvidia not found in directory /lib/modules/5.10.104-tegra
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: HD Pro Webcam C920
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"HD\ Pro\ Webcam\ C920", v4l2.device.bus_info=(string)usb-3610000.xhci-4.2, v4l2.device.version=(uint)330344, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 53 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1536, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1296, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/2, 5/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 10/1, 15/2, 5/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [11] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [12] video/x-raw, format=(string)YUY2, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [13] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [14] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [15] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [16] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [17] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [18] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)90, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [19] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [20] image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [21] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [22] image/jpeg, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [23] image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [24] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [25] image/jpeg, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[gstreamer] [26] image/jpeg, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
^Creceived SIGINT
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
video-viewer:  shutdown complete

malcolmcapel, 2023-02-11:
on the main system (outside of the jetson-inference container) there is an nvdia.ko on /lib/modules/5.10.104-tegra/extra/opensrc-disp

dusty-nv, 2023-02-11:
I get that message too, but my Logitech C920 works okay with the same command.    Just for debugging purposes, could you try setting `--input-codec=raw` ?

You could try mounting that `/lib/modules/5.10.104-tegra` directory into the container to see if it makes any difference...

malcolmcapel, 2023-02-11:
Dustin

Would you mind telling me how to mount /lib/modules/5.10.104-tegra into the container? I’m a total newbee with respect to docker.

Thanks,
M

dusty-nv, 2023-02-11:
I just tried doing `docker/run.sh -v /lib/modules/5.10.104-tegra:/lib/modules/5.10.104-tegra` and that got rid of that error message.  It didn't change anything else about the way it ran for me though.

malcolmcapel, 2023-02-11:
Dustin

--input-codec=raw worked in getting video-viewer to work under the jetson-inference container.  I had also invoked the docker container with the option "v /lib/modules/5.10.104-tegra:/lib/modules/5.10.104-tegra".  If I leave off the input-codec option then video-viewer fails as before.

M

malcolmcapel, 2023-02-11:
bye the way --input-codec=mjpeg also worked for this webcam  (Logitech HD Pro C920 )

dusty-nv, 2023-02-12:
Hmm okay, thanks.  For some reason my C920 only has raw and mjpeg modes, no h264.  I guess h264 was added in a later revision of the camera?

closed 2023-04-27 by dusty-nv