dusty-nv/jetson-inference #1559: camera-capture:  failed to capture RGBA image from camera

mohitg0292, 2023-02-11:
Hi!
I’m using NVDIA Jetson nano 4GB kit with Jetpack 4.6.3.
I successfully built the project from source. And I’ve connected Raspberry Pi camera too.

When I’m trying to collect the data from camera by running below command, I’m getting an error.

`ttz@ttz-desktop:~$ camera-capture csi://0 

[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw(memory:NVMM) ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------

camera-capture:  successfully initialized video device (1280x720)
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1290x730
[OpenGL] glDisplay -- display device initialized (1290x730)
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
[gstreamer] gstreamer message stream-start ==> pipeline0
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:751 Failed to create CaptureSession
[gstreamer] gstCamera -- end of stream (EOS)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstreamer pipeline0 recieved EOS signal...
[gstreamer] gstDecoder -- failed to retrieve next image buffer
camera-capture:  failed to capture RGBA image from camera
[gstreamer] gstDecoder -- failed to retrieve next image buffer
camera-capture:  failed to capture RGBA image from camera
[gstreamer] gstDecoder -- failed to retrieve next image buffer
camera-capture:  failed to capture RGBA image from camera
[gstreamer] gstDecoder -- failed to retrieve next image buffer
camera-capture:  failed to capture RGBA image from camera
[gstreamer] gstDecoder -- failed to retrieve next image buffer
camera-capture:  failed to capture RGBA image from camera
`

The camera is working perfect while running the “Hello camera” project in Nvdia basic course.

Can you please help what can be the reason?

Regards
Mohit Gupta

mohitg0292, 2023-02-11:
@dusty-nv can you please suggest a solution?
thanks

mohitg0292, 2023-02-11:
I tried running this solution too but getting same error.

_cd /path/to/your/jetson-inference/build
cmake -DENABLE_NVMM=OFF ../
make
sudo make install_

Also when I ran "nvgstcapture-1.0", I got this
[image]

dusty-nv, 2023-02-12:
Hi Mohit, if nvgstcapture also fails to display the capture, then there may be something wrong with the camera connection, or the nvargusdaemon needs restarted.   You can do that with:

sudo service nvargus-daemon restart

[image]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F101650936%2F218271558-2d194bab-1309-4642-9ff4-bc6041977f10.png&data=05%7C01%7Cdustinf%40nvidia.com%7Cd8b1acb7cc3d428acfe408db0c539a12%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C638117325430044958%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=030GO1COQPwjMsaH5RQ2QVdqGblydasWuuvvxygYrAY%3D&reserved=0>

mohitg0292, 2023-02-12:
I tried changing camera to another Raspberry Pi v2 and it worked. 

I'm not sure why previous camera was working with basic "Hello Camera" and not with this module. 

But thanks for replying.

dusty-nv, 2023-02-12:
OK gotcha.  Regardless, glad you got it working!  👍