dusty-nv/jetson-inference #1560: Jetson Nano - 'jetson-inference' module not found

DrKornballer, 2023-02-11:
I know its been a while, but after many attempts I can't get the newly trained dataset to work. Running newest vs of Jetpack, upgraded to Ubuntu 20.04, upgraded Pytorch and Torchvision and got the training to work, but now when I run:

imagenet.py --model=$NET/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/dog/01.jpg dog.jpg

I get:

Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 27, in <module>
    from jetson_inference import imageNet
ModuleNotFoundError: No module named 'jetson_inference'

Any suggestions?

dusty-nv, 2023-02-12:
Hi there, what Jetson are you using?

If you upgraded Ubuntu distro, I'd recommend re-compiling/re-installing jetson-inference from source because the Python version may have changed from 3.6 to 3.8

closed 2023-04-27 by dusty-nv

vnguyen13, 2023-06-15:
I'm having this same error. Here's where I'm at:
1.  Started with a 128GB microSD card
2. Flashed it with Jetpack 4.5
3. Built the jetbot package (from the docker version)
4. Built the jetson-inference from code

From there, I tried running:
`root@jetbot-cmra-01:/workspace/jetson-inference/build/aarch64/bin# ./imagenet.py /workspace/jetson-inference/data/images/orange_0.jpg /workspace/jetson-inference/data/images/test/output_0.jpg`

And got:
`Traceback (most recent call last):
  File "./imagenet.py", line 27, in <module>
    from jetson_inference import imageNet
ModuleNotFoundError: No module named 'jetson_inference'`

@dusty-nv Any ideas? Thanks!

dusty-nv, 2023-06-15:
@vnguyen13 are you able to run `python3 -c 'import jetson.inference'` ?

Is this running in the jetbot container (from github.com/NVIDIA-AI-IOT/jetbot) ?

Do you have a Dockerfile for how you built jetson-inference in it?

vnguyen13, 2023-06-20:
@dusty-nv when I run that within the Jupyter Notebooks terminal, this is what I get:

> root@jetbot-cmra-01:/workspace# python3 -c 'import jetson.inference'
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> ModuleNotFoundError: No module named 'jetson'

When I SSH in as the jetbot user, I get:

> jetbot@jetbot-cmra-01:~$ python3 -c 'import jetson.inference'
> warning:  importing jetson.utils is deprecated.  please 'import jetson_utils' instead.
> warning:  importing jetson.inference is deprecated.  please 'import jetson_inference' instead.

For the jetson-inference project, I used these instructions:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

vnguyen13, 2023-06-20:
@dusty-nv I got it working. I removed the jetson-inference folder and tried again but going through these directions:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

I'm able to run the imagenet command (`./imagenet images/jellyfish.jpg images/test/jellyfish.jpg`) on the Jupyter Notebook terminal.