dusty-nv/jetson-inference #1562: Detectnet object detection two classes using different confidence values.

NigelPG, 2023-02-13:
Hi Dusty.
I've been using the Jetsaon Nano for about two years now and have a problem I cant solve ;)
I have trained SSD to discriminate between plates and scratched palates (classes are plate, scratch).
I set the confidence at 0.5 and it just about works but misses some scratches. If I set it at 0.4 it picks up all the sctaches but sometimes sees flase positives for plates. What i want to do is use diffent confidence vales for ewach of the two classes. I realise I cant do this at the initial level but though i might be able to set the confidence at 0.4 then extract the actual confidence value for the each object detected.
How can I do do that? 
I curretly use the class detection to set gpio so i have that part sorted, just need to get the confidence with the class.
Please help, best regards.
Nigel

dusty-nv, 2023-02-13:
Hi @NigelPG, I think you are right that the easiest way to do it would be to just check the confidence values after detection.   Are you using Python or C++?   Here are the class members for the `detectNet.Detection` object:

https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#detectNet

``` 
Detection = <type 'jetson.inference.detectNet.Detection'>
Object Detection Result
 
----------------------------------------------------------------------
Data descriptors defined here:
 
Area
    Area of bounding box
 
Bottom
    Bottom bounding box coordinate
 
Center
    Center (x,y) coordinate of bounding box
 
ClassID
    Class index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Confidence
    Confidence value of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Height
    Height of bounding box
 
Instance
    Instance index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Left
    Left bounding box coordinate
 
Right
    Right bounding box coordinate
 
Top
    Top bounding box coordinate
 
Width
     Width of bounding box
```

So you would just look at the `detection.Confidence` member like you are currently doing with `ClassID`

NigelPG, 2023-02-14:
Thanks for the quick response Dusty, I'll give that a go and get back with how it goes.

NigelPG, 2023-02-14:
Fantastic, I got it to work once I realised I didn't need to call net.detect or net.whatever.
If anyone needs a quick solution, this will let you set the overall threshold in the usual way but only act when specific thresholds for classes are met.
	while True:
		img = camera.Capture()
		detections = net.Detect(img)
		display.Render(img)
		display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
	
		for detection in detections:
			class_name = net.GetClassDesc(detection.ClassID)
			conf = (detection.Confidence)
			if class_name == "Plate":
				print ("plate")
				print (conf)
				if conf > 0.9:
					GPIO.output(output_pin2, GPIO.HIGH)
					timer_p2 = time.time()		
			if class_name == "Scratch":
				print ("scratch")
				GPIO.output(output_pin1, GPIO.HIGH)
				timer_p1 = time.time()
		if time.time() - timer_p1 > 0.5:
			GPIO.output(output_pin1, GPIO.LOW)
		if time.time() - timer_p2 > 0.5: 
			GPIO.output(output_pin2, GPIO.LOW)

closed 2023-04-27 by dusty-nv