dusty-nv/jetson-inference #1563: Not able to install Pytorch

mohitg0292, 2023-02-16:
Hi!

I installed Jetpack 4.3 on nvdia jetson nano. While building the project from scratch I skipped Pytorch installation. But now when I tried to train the model using below command.
python3 train.py --model-dir=models/<YOUR-MODEL> data/<YOUR-DATASET>

I got an error saying "Module torch not found". Then I tried to install pytorch explicitly by using command ./install-pytorch.sh

I got only one option
[image]

I'm trying to install it but the logs are never ending but multiple things. I'm attaching few logs here.

Please help to resolve the issue. 
PS: These are the python versions present currently in system.

[image]

Thanks
Mohit

dusty-nv, 2023-02-16:
> I'm trying to install it but the logs are never ending but multiple things. I'm attaching few logs here.

Hi @mohitg0292, it can take a while for Nano to build torchvision - let it run for a while until it completes.  Or you could run the jetson-inference docker container, which already has PyTorch/ect installed in it for you:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md

mohitg0292, 2023-02-16:
> Hi @mohitg0292, it can take a while for Nano to build torchvision - let it run for a while until it completes. Or you could run the jetson-inference docker container, which already has PyTorch/ect [...]

Ok. I'll try to run the command again and will leave it for while. Just wanted to make sure the attached logs don't have any errors?

dusty-nv, 2023-02-16:
No they don't appear to, it's just compiling

mohitg0292, 2023-02-16:
You were right. It took 35 minutes to complete installation.

Now when I'm training the model, getting this error.
[image]

dusty-nv, 2023-02-17:
@mohitg0292 can you try running `pip3 install 'pillow<9'`

mohitg0292, 2023-02-17:
thanks. It worked.

One question. I trained model for 1 epoch and tested the model using.
$ imagenet.py --model=models/<YOUR-MODEL>/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=data/<YOUR-DATASET>/labels.txt csi://0

On the camera screen, it is showing me "100% Class#0" and not recognizing the object when I put the object in front of camera.

Is there anything I'm missing?

mohitg0292, 2023-02-17:
ok. so basically it will not work for one object?
and what does it mean which is displaying on camera screen. "100% Class#0"?

dusty-nv, 2023-02-17:
There should be a minimum of two classes, yea

mohitg0292, 2023-02-17:
I just trained the model for 6 objects again after deleting .engine file as you suggested. When I tested, it is again showing me class#0 or class #1 or class#4 or class #5, not the name of objects. 
also i checked if it is showing the serial number of object but that's also not the case. It is also showing same class for two objects.

Is there anything i'm still missing?

Thanks for your quick response. :)

dusty-nv, 2023-02-17:
Do you have a labels.txt file with your class names that you are specifying on the command line when you launch the program?

mohitg0292, 2023-02-17:
The lablels.txt file looks like this
[image]

This is the output (this one is second item in labels file):
[image]

mohitg0292, 2023-02-21:
@dusty-nv Can you please look into this issue? 
I'm really stuck in this.

Thanks
Mohit

dusty-nv, 2023-02-21:
What's the command-line that you are running imagenet program with?    When it starts, it should print out something like `"loaded 4 class labels"`

mohitg0292, 2023-02-21:
Thanks for your quick reply. 
I just tested the model again and noticed that it wasn't able to load the labels because of wrong path mentioned in the command. Finally I corrected the path and the labels are loaded. But the accuracy isn't good. and it is not showing the correct object name.

However I've also included "hand" and "background" in labels and retraining the model. I'll again test once the retraining will be finished. 

On what the accuracy depends? The number of objects or the background light also (you can see the background light in above pics)?
Also while clicking pictures, I didn't have green background like you showed in the video so I just put green paper in the back for "val". And I put a colored book in the background for "test" pics. Is it fine?

Mohit

mohitg0292, 2023-02-21:
hi @dusty-nv !
Ok. I just finished training the model and it was working to identify the object. As the next step I tried to configure the audio and made the changes as suggested on the description page.

I also made changes in imagenet.py but when i tried to run it, it is showing this error.
[image]
Then I restarted the JEtson and tried again then it showed this but camera window didn't open.
[image]

Also when i undo the changes and tried to run the original imagenet.py it showed me same capturesession error.

Not sure what I did wrong.

Is there something I'm missing? Do we need to do something after making changes in imagenet.py? (like compiling something or exporting onnx file etc?)

I'm not sure why everything stopped working suddenly.

Please help
Mohit

dusty-nv, 2023-02-22:
> is there something I'm missing? Do we need to do something after making changes in imagenet.py?

I would recommend making your own copy of imagenet.py and putting it somewhere else, or making changes to `jetson-inference/python/examples/imagenet.py` (and then running `cmake ../ && make && sudo make install` from the build dir so the updated imagenet.py gets installed)

If you think a problem with your camera has occurred, try running video-viewer.py first to confirm the camera is still working.

mohitg0292, 2023-02-22:
Hi @dusty-nv !

As you suggested I followed below steps:
- created a separate file in jetson-inference/python/examples/food_container_identifier.py and kept imagenet.py as it is. 
- Ran "cmake ../ && make && sudo make install"
- Trained the model using 
python3 train.py --model-dir=models/food_container_identifier data/food_container_identifier
- Exported model in Onnx format using
python3 onnx_export.py --model-dir=models/food_container_identifier
- Then tested the model using
imagenet.py --model=/home/ttz/jetson-inference/python/training/classification/models/food_container_identifier/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=/home/ttz/jetson-inference/python/training/classification/data/food_container_identifier/labels.txt csi://0

The camera is working at this stage.

-----------------------Next step--------------------------
Then I ran below command to check the audio.
python3 /home/ttz/jetson-inference/python/examples/food_container_identifier.py --model=/home/ttz/jetson-inference/python/training/classification/models/food_container_identifier/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=/home/ttz/jetson-inference/python/training/classification/data/food_container_identifier/labels.txt csi://0

In this case the audio is working and showing this output but the camera screen is not displayed and the system is just repeating one object name and the logs are like this.
[image]

After this when I ran imagenet again, I got this error: 
[image]

To get rid of this error I restarted the Jetson nano and ran imagenet. The camera is working fine. Then I ran food_container_identifier.py to check audio. Again the camera screen didn't work and I was just getting audio. Then when I ran imagenet.pu the camera error is displayed.

So that's a pattern I found out. I hope I was able to explain it.

Thanks
Mohit

mohitg0292, 2023-02-22:
@dusty-nv I copied and pasted the whole code from here.
https://github.com/oliver-almaraz/food_container_identifier/blob/main/food_container_identifier.py

I didnt make any changes in existing code.

dusty-nv, 2023-02-22:
OK gotcha, thanks - the camera screen isn't displaying because you have commented out the videoOutput stuff.  

I may be the audio should run in a different thread - I'm not sure if the audio call to `engine.runAndWait()` blocks and takes extra time or not.

Other than an ungraceful shutdown, I don't know why the CSI camera doesn't restart.  However you should be able to run `sudo systemctl restart nvargus-daemon` instead of rebooting your system to get it back.

mohitg0292, 2023-02-22:
Ok. But in the practical example we do need the video output, right? Also in your video tutorial on youtube you have also showed the same. Why do we need to comment the video output then?

Ok. I'll try to run the command rather than restarting the system. 

But what's the solution I can opt for? Should I send you the complete logs again?

dusty-nv, 2023-02-22:
You should be able to leave the videoOutput uncommented, sure.  The videoOutput can save the results to video or a network stream if you prefer over a physical display.

mohitg0292, 2023-02-22:
@dusty-nv ok I uncommented videoOutput and other related variables and the video screen is displayed now.
But there are some issues:
- Since i'm using separate screen using HDMI port, the videooutput is very slow with audio using `food_container_identifier.py`.
- I did some hit and trial. What I found is after running `food_container_identifier.py`, I've to run the command `systemctl restart nvargus-daemon` otherwise I get error `failed to create capture session` in running `imagenet.py`
- Today I trained the model again and now the objects are not recognized correctly. Not sure what happened differently from yesterday. It was all good in training and running the model using `imagenet.py` yesterday.

Could you please share which screen you used while making this video?
https://www.youtube.com/watch?v=PH_wht8jTyA
In this sample, the videoOutput it is quite smooth.

Thanks
Mohit

dusty-nv, 2023-02-22:
> * Since i'm using separate screen using HDMI port, the videooutput is very slow with audio using `food_container_identifier.py`.

I think it may be slowed down by the blocking call to audio `engine.runAndWait()`   
Try commenting that out and seeing if it speeds up
I use a HDMI screen and hauppage device to capture it for youtube videos

mohitg0292, 2023-03-09:
Hi @dusty-nv !
I have again trained the program for around 35 objects. The Jetson is not recognizing the objects correctly.
In addition to that I have a query regarding the audio output. Currently I'm using a separate screen using HDMI port and using the audio from the screen.

Now I want to test it without the screen, If I understand it correctly: 
I'll connect the jetson with usb using my laptop and run the program through command prompt. I'll probably connect bluetooth speakers via bluetooth device on jetson. Or using the USB to 3.5 mm jack converter and connecting my earphones to 3.5mm jack.
Then I'll run the program through my laptop. The camera and audio output should run right?

Thanks
Mohit

dusty-nv, 2023-03-09:
@mohitg0292 yes they should still run without display, but if you want to view the camera from your laptop you'll need to stream it over RTP

mohitg0292, 2023-03-17:
Hi @dusty-nv. Thanks, I have configured the bluetooth speaker successfully. But the model is not able to identify any object. The audio is saying only two objects i.e. "Dairy 50g", "tea 200g" and nothing else. I tried to put different objects in front of the camera other than "Dairy 50g", "tea 200g"  but no success. Even the "hand" and "background" are also not identified. The audio also doesn't say "Dove Shampoo 200 ml" and "Butter 350g " which is shown in the output below.

Output on terminal.
class 0005 - 0.403012  (Dairy 50g)
class 0025 - 0.595613  (tea 200g)
class 0005 - 0.450656  (Dairy 50g)
class 0025 - 0.547858  (tea 200g)
class 0005 - 0.313710  (Dairy 50g)
class 0025 - 0.685206  (tea 200g)
class 0005 - 0.410517  (Dairy 50g)
class 0025 - 0.588039  (tea 200g)
class 0005 - 0.473894  (Dairy 50g)
class 0025 - 0.524553  (tea 200g)
class 0005 - 0.406709  (Dairy 50g)
class 0025 - 0.591758  (tea 200g)
class 0005 - 0.425662  (Dairy 50g)
class 0025 - 0.572790  (tea 200g)
class 0005 - 0.393630  (Dairy 50g)
class 0025 - 0.604921  (tea 200g)
class 0005 - 0.353506  (Dairy 50g)
class 0025 - 0.645136  (tea 200g)
class 0025 - 0.999858  (tea 200g)
class 0025 - 0.999902  (tea 200g)
class 0025 - 0.999969  (tea 200g)
class 0025 - 0.998161  (tea 200g)
class 0005 - 0.479680  (Dairy 50g)
class 0025 - 0.518658  (tea 200g)
class 0005 - 0.408632  (Dairy 50g)
class 0025 - 0.589930  (tea 200g)
class 0005 - 0.401045  (Dairy 50g)
class 0025 - 0.597354  (tea 200g)
class 0005 - 0.076208  (Dairy 50g)
class 0025 - 0.921183  (tea 200g)
class 0005 - 0.903241  (Dairy 50g)
class 0025 - 0.095948  (tea 200g)
class 0025 - 0.999941  (tea 200g)
class 0025 - 0.999964  (tea 200g)
class 0025 - 0.999851  (tea 200g)
class 0025 - 0.999095  (tea 200g)
class 0025 - 0.996700  (tea 200g)
class 0025 - 0.995081  (tea 200g)
class 0005 - 0.014279  (Dairy 50g)
class 0025 - 0.985547  (tea 200g)
class 0005 - 0.886126  (Dairy 50g)
class 0008 - 0.034091  (Dove Shampoo 200 ml)
class 0025 - 0.071053  (tea 200g)
class 0005 - 0.987528  (Dairy 50g)
class 0025 - 0.999926  (tea 200g)
class 0005 - 0.278210  (Dairy 50g)
class 0025 - 0.721621  (tea 200g)
class 0005 - 0.189403  (Dairy 50g)
class 0025 - 0.809976  (tea 200g)
class 0005 - 0.999054  (Dairy 50g)
class 0005 - 0.999771  (Dairy 50g)
class 0005 - 0.999645  (Dairy 50g)
class 0005 - 0.997947  (Dairy 50g)
class 0025 - 0.999884  (tea 200g)
class 0005 - 0.980071  (Dairy 50g)
class 0025 - 0.015116  (tea 200g)
class 0025 - 0.999876  (tea 200g)
class 0025 - 0.999899  (tea 200g)
class 0025 - 0.999871  (tea 200g)
class 0025 - 0.999880  (tea 200g)
class 0025 - 0.999848  (tea 200g)
class 0025 - 0.999883  (tea 200g)
class 0025 - 0.996988  (tea 200g)
class 0025 - 0.998979  (tea 200g)
class 0003 - 0.039865  (Butter 350g)
class 0005 - 0.030806  (Dairy 50g)
class 0025 - 0.928849  (tea 200g)
[... 68 lines cut ...]
class 0025 - 0.999880  (tea 200g)
class 0025 - 0.999848  (tea 200g)
class 0025 - 0.999883  (tea 200g)
class 0025 - 0.996988  (tea 200g)
class 0025 - 0.998979  (tea 200g)
class 0003 - 0.039865  (Butter 350g)
class 0005 - 0.030806  (Dairy 50g)
class 0025 - 0.928849  (tea 200g)
class 0005 - 0.998429  (Dairy 50g)
class 0005 - 0.071530  (Dairy 50g)
class 0025 - 0.927616  (tea 200g)
class 0005 - 0.481068  (Dairy 50g)
class 0025 - 0.516110  (tea 200g)
class 0005 - 0.878848  (Dairy 50g)
class 0025 - 0.118939  (tea 200g)
class 0025 - 0.999926  (tea 200g)

I have a few more questions. Please answer them.
1. How to train the model so that it is able to recognize the objects in any environment whether it is supermarket or home or kitchen etc.?
2. Can we take pictures from mobile for any object and then train the model? Or is it that the pics have to be taken from the camera attached to the jetson (I'm using raspberry Pi v2 camera)?
3. I have also uploaded the pics on the [Google drive](https://drive.google.com/drive/folders/18eYWHip2yn6OOU00qHzNcJYm7L8yt6wK?usp=sharing) which i'm using to train the model. Can you briefly have a look at them and check I'm using the right method? (I tried to upload them on github but getting errors due to the huge size)
4. I have also noticed that once you train the model and immediately run it then it is able to give different names and even recognize some of them but after a few days something happens and it just says 2-3 names irrespective of any object. I know it doesn't make any sense but I just noticed it.

Thanks a ton
Mohit

dusty-nv, 2023-03-17:
> 3\. I have also uploaded the pics on the [Google drive](https://drive.google.com/drive/folders/18eYWHip2yn6OOU00qHzNcJYm7L8yt6wK?usp=sharing) which i'm using to train the model. Can you briefly have [...]

Hi @mohitg0292, from browsing through your dataset, I noticed that a bunch of the classes in the `train/` folder don't have any images in them.  So the model would be untrained in those classes.   You should make sure that every class has a bunch of images for it (ideally 100 or more).   A lot of times, to save time collecting the dataset, I use all the images I have for `train/` and just copy them to `test/` and `val/`  (that's certainly not a best-practice for making a real model, as the test/val sets should be independent, but when you are just in development trying to get the model to work, it can save you time)

> 1. How to train the model so that it is able to recognize the objects in any environment whether it is supermarket or home or kitchen etc.?

When in doubt, add more data to your dataset - with a variety of backgrounds, camera angles, lighting conditions, ect.   The number of images for each class should be relatively balanced.

> Can we take pictures from mobile for any object and then train the model? Or is it that the pics have to be taken from the camera attached to the jetson (I'm using raspberry Pi v2 camera)?

Yea, you can take pictures with any camera, and then simply upload them or copy/paste them into your appropriate dataset folders later

mohitg0292, 2023-03-17:
Hi @dusty-nv ! Thanks for your quick reply.
sorry i forgot to mention. actually i uploaded few pics on drive, not all of them since there were around 4000 images and it was taking too much time. 
But all folders (train, text, val) have images with respective objects in the model. But there has to be some logic that the model only identifies these 2 objects (tea, dairy). or not?

I'm not sure but even "hand" or "background" is not identified which is pretty simple.

dusty-nv, 2023-03-17:
Do those tea and dairy objects have a lot more images than the others?   What accuracy is your model training to?

You might want to add additional code to train.py to make sure the dataset is loading all the images.  I noticed that your folder names have spaces in them, I don't think that should affect it but I'm not sure.

mohitg0292, 2023-03-17:
No every object has same number of images. Like you mentioned in your video. (100 images for training, 20 for val and 5-10 for test). I used blank wall for training images, then put green paper background for val images and colored background for test images. Also sometimes it's different object than tea and dairy. Like "ketchup" repeatedly...
The folders have spaces because as you mentioned in labels.txt file, I have put names in same format. The model is building the folders automatically. 
If I'm not wrong, we call train.py to train the model right? ITt doesn't start if I run the "food_container_identifier.py" script/

`What accuracy is your model training to?` Where can I find this?

I just ran the model again. This time I didn't even put any object in front of camera. Just my hand in different positions. But the audio just said "tea" repeatedly. Only very few times it said, Dermaglos, zucker that's it.
 
I'm attaching full logs here.
[logs.txt](https://github.com/dusty-nv/jetson-inference/files/11003436/logs.txt)

dusty-nv, 2023-03-17:
The accuracy is reported when you run train.py as you train it.  I believe you can also run it with the --eval flag to just do the validation and skip training

Have you been deleting the .engine file when you re-export the ONNX?

mohitg0292, 2023-03-17:
@dusty-nv Thanks for your quick responses. 
I just tried to run train.py again with --eval parameter. But the model is running to train the system and didn't skip training. 
Yes I always delete .engine file as you recommended.

Can we use AI generated image to train the model? Also would you suggest to add more pictures only in train folder or all the folders? As I didn't really understand the purpose of two other folders i.e. "val" and "test".

Mohit

mohitg0292, 2023-03-20:
Hi @dusty-nv !
As you suggested I took more pictures from my mobile camera and put it in train, val and test folders of respective object in ratio 70:20:10. 
Then I trained the model after deleting .engine file. It tool around 7 hours to train the model. Here's the [training output](https://docs.google.com/document/d/1TXsc4K8B4GGgpACuTI3wQOrJw8cVfp6wa8h-t3GNgpU/edit) of training the model.
After that I exported in .onnx format. here's [onnx output](https://docs.google.com/document/d/1dE9pJwnWgR2Vp-KW8b8Ly3XPzonTrCf15eYIuPKQZEY/edit) of onnx.

The labels.txt look like this.
[image]

I ran the model and no success. The model is totally recognizing wrong now. :(
Here are logs for the running the model.
[model running logs.txt](https://github.com/dusty-nv/jetson-inference/files/11012684/logs1.txt)

Please help and guide me where I'm going wrong. Also in the logs if you'll notice it is printing whole labels in the output.

Mohit

dusty-nv, 2023-03-20:
OK, perhaps this because of how you have named your folders and the alphanumeric order in which PyTorch reads them, being different from your labels.txt.   In your training log, this is the order of the classes that the model was trained with:

``` bash
=> dataset classes:  27 ['Amchur Powder 100g', 'Butter 350g', 'ChainSpray 200ml', 'Dairy 50g', 'Dove Shampoo 200 ml', 'Fastrack ', 'Garam Masala 100g', 'Gefro Box 500g', 'Handwash 500ml', 'Incense 50g', 'Lemon 200ml', 'Mate 500g', 'Pepper 50g', 'Pepper Punt 50g', 'Ponds 90g', 'Salz 500g', 'SpecsBox 100g', 'Thymin 50g', 'Zucker 500g', 'background', 'deo 150ml', 'dermaglos 50g', 'hand', 'ketchup 500ml', 'lechuga crema 55g', 'tea 200g', 'vaseline 100ml']    
```

Try changing your labels.txt to reflect this same order, and see if that helps.

mohitg0292, 2023-03-26:
Hi @dusty-nv !!

Thanks for your feedback. You were correct. The model was fetching wrong folders with wrong properties. Actually I didn't create folders manually. The camera-capture tool automatically created folders based on the labels.txt. So for now I manually removed all the spaces from all the folders. But I do have few doubts:
1. How to disable camera-capture tool creating folders by default? For ex. I added few more objects then again camera-capture tool created folders again repeatedly (since the labels file have spaces and existing folders don't have spaces) and I had to delete the new empty folders.
2. Can we train the model only for new objects or do we need to train the whole model again? 
3. I have run the --eval command. Can you tell me the performance of the model based on the output? I didn't get it.
[image]

4. While training the model it runs for train (774) and test (176). What are these numbers?
5. The output of the model is slow. Can you check the code and tell me if it can be improved?
https://drive.google.com/file/d/1ItMTk16HcKsTg3nNDoTbkHZ7H4lsTEKI/view?usp=sharing

Thanks
Mohit Gupta

dusty-nv, 2023-03-26:
> 1. How to disable camera-capture tool creating folders by default? For ex. I added few more objects then again camera-capture tool created folders again repeatedly (since the labels file have spaces and existing folders don't have spaces) and I had to delete the new empty folders.

camera-capture knows where to put the images based on the labels.txt, so your labels should match the folder names.  Or, after your done training, you can just have a separate labels.txt that has spaces in it.

> 2\. Can we train the model only for new objects or do we need to train the whole model again?

If you keep your new objects in a separate dataset, you should be able to load your previous model from train.py and continue on from there.

> 3\. I have run the --eval command. Can you tell me the performance of the model based on the output?

It says the accuracy (`Acc@1`) is only 3.915% - this doesn't seem correct.  Maybe you loaded a model that wasn't trained for many epochs.  Regardless, it also outputs the accuracy at the end of each training epoch.

> 4\. While training the model it runs for train (774) and test (176). What are these numbers?

Those are the batch numbers.   If you multiple the batch number by the batch size, that gives you the number of images in the dataset

> 5\. The output of the model is slow. Can you check the code and tell me if it can be improved?
>     https://drive.google.com/file/d/1ItMTk16HcKsTg3nNDoTbkHZ7H4lsTEKI/view?usp=sharing

I think the calls to pyttsx in your main processing loop are blocking and may be slowing it down:

```
engine.say(class_desc)
engine.runAndWait()
```

those should probably be run in a separate thread, so the camera processing loop can continue on

closed 2023-04-27 by dusty-nv