dusty-nv/jetson-inference #1565: How  to get raw Detection results for custom model

sezer-muhammed, 2023-02-19:
Hi, 
I want to stream on RTP, save a video file and use DNN model on webcam video. Memory and performance is limited so I first tried jetson_inference library for all of these jobs.

But ssd-mobilenet or other models are not good enough. On  the other hand using opencv and traditional codes not effecient as jetson_inference. 

I need to use yolov8n as model, stream on RTP, save to file and draw some lines on the image. Any alternatives to my problem?

Besides asking for an alternative suggestion, my main question is:

```
        self.plane_detection_model = jetson.inference.detectNet(
            "ssd-mobilenet-v2",
            threshold=0.4,
            argv=[
                '--model=' + DETECTION_MODEL_PATH, '--labels=' + LABELS_PATH,
                '--input-blob=images', '--output-cvg=output0',
                '--output-bbox=output0'
            ]
        )
```

This is how I declared my network.

[image]
This is my networks onnx results.

How can I obtain this output0 layers results direclty? And How can I decide model input resolution? Because resolution seems is 300x300. I need to change it but I could not find anything for that in the code.

dusty-nv, 2023-02-19:
You would need to update the pre/post-processing in `c/detectNet.cpp` to support YOLOv8 model.   

You might be interested in this project which appears to already support YOLOv8 + TensorRT:  https://github.com/triple-Mu/YOLOv8-TensorRT

sezer-muhammed, 2023-02-20:
Hi, 

I have some questions to understand situation better:

Can I print some data in detecnet.cpp? İf so with whic command? Cout, printf, logerror?

Can we get image data without converting it to numpy array? And feed the model?

I do not want to use any other image data other than jetson inference image object. What can I do for this task?
Yolo + stream + fila save at high Fps?

Thanks

dusty-nv, 2023-02-20:
> Can I print some data in detecnet.cpp? İf so with whic command? Cout, printf, logerror?

Sure, you can use any of those commands for printing to the console.  For just some quick temporary text, I'd probably just use printf.  If I intended for it to be permanent, I would use LogVerbose/LogDebug/ect

> Can we get image data without converting it to numpy array? And feed the model?

Do you mean that you have image data in Python that you want to get into CUDA and feed the model with?    Or that you want to access the CUDA image memory without converting it to numpy?

sezer-muhammed, 2023-02-21:
Hi Dusty!,

I have one last question. I guess label 0 named "BACKGROUND" is not count as detection. But in the repo that you suggested returns only the detection with highest score (1, N). This means on the jetson inference side detection label is always 0 since ID with 0 has the highest score. 

I am lucky since I have only one object for model, so I just need to remove BACKGROUND from the detect.cpp but I could not find where is it. Could you help me?

sezer-muhammed, 2023-02-21:
Got it.

[image]
30 FPS with yolov8, on ros2, stream on RTP, file saving.

Thanks!

Note: I'll try to set a guide for those who need. If you need just let me know

closed 2023-02-21 by sezer-muhammed