dusty-nv/jetson-inference #1566: Not able to load .onnx files in detectnet.py

SaTz1483, 2023-02-20:
Hey @dusty-nv how do I use a custom model i trained on detection algorithm.
When i give my model .onnx path as input to the argument it throws me this error.

[image]

The output,

[image]

dusty-nv, 2023-02-20:
Hi @SaTz1483, it looks like you need to set the other command-line parameters for loading custom model, like the following:

``` bash
detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes
```

Also it says it can't find the model file that you specified, so double-check it exists.

SaTz1483, 2023-02-28:
@dusty-nv I'm actually trying to train a custom dataset that has a resolution of 4208x3120 and its taking ridiculously long. Is there any way of optimizing this? 
Does resolution play a role in training time?

SaTz1483, 2023-02-28:
Hey @dusty-nv
Could you help me out with this?

 I'm actually trying to train a custom dataset that has a resolution of 4208x3120 and its taking ridiculously long. Is there any way of optimizing this? 
Does resolution play a role in training time?

closed 2023-04-27 by dusty-nv