dusty-nv/jetson-inference #1573: Cannot find -lnpymath when building for Python 3.8

Aldhanekaa, 2023-03-04:
I try to build jetson inference in my Jetson Nano (2GB) with python3.8, because it's requirement for my project. 
I follow instruction from https://github.com/dusty-nv/jetson-inference/issues/936 , by adding 3.8 in CMakeList.txt, however when I try to make (`make -j$(nproc)`), an error occured) .

 ```

79%] Built target camera-capture_autogen
[ 84%] Built target jetson-inference-python-36
Scanning dependencies of target jetson-inference-python-37
/usr/bin/ld: cannot find -lnpymath
collect2: error: ld returned 1 exit status
utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/build.make:283: recipe for target 'aarch64/lib/python/3.8/jetson_utils_python.so' failed
make[2]: *** [aarch64/lib/python/3.8/jetson_utils_python.so] Error 1
CMakeFiles/Makefile2:1243: recipe for target 'utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/all' failed
make[1]: *** [utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

---

Numpy is already installed on my Python3.8, with version `1.23.2`

dusty-nv, 2023-03-04:
@Aldhanekaa I haven't tried building jetson-inference with Python 3.8 on JetPack 4.x, but you will need to have the dev libraries installed for numpy (normally I do `apt-get install python3-numpy`), or you may need to make sure that cmake finds them

Aldhanekaa, 2023-03-04:
> @Aldhanekaa I haven't tried building jetson-inference with Python 3.8 on JetPack 4.x, but you will need to have the dev libraries installed for numpy (normally I do `apt-get install python3-numpy`), or you may need to make sure that cmake finds them

ah I see. The error is resolved now by installing numpy, thanks for that. However new error occured;
```
/usr/bin/ld: /usr/local/lib/libpython3.8.a(abstract.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_Py_add_one_to_index_F' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(abstract.o): In function `PyBuffer_FromContiguous':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/abstract.c:529:(.text+0x174c): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(abstract.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_Py_add_one_to_index_C' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/abstract.c:529:(.text+0x1750): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(object.o): In function `_PyObject_GetMethod':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:1143:(.text+0x2494): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(object.o): In function `_PyObject_LookupAttr':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:961:(.text+0x2b30): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:961:(.text+0x2c78): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:961:(.text+0x2db0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:961:(.text+0x2ef0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/object.c:961:(.text+0x3040): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(typeobject.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyType_GenericAlloc' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(typeobject.o): In function `type_new':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/typeobject.c:2782:(.text+0x100f0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(unicodectype.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_PyUnicode_TypeRecords' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(unicodectype.o): In function `gettyperecord':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/unicodectype.c:55:(.text+0x6404): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(unicodectype.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_PyUnicode_ExtendedCase' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(unicodectype.o): In function `_PyUnicode_ToTitlecase':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/unicodectype.c:69:(.text+0x6428): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(unicodectype.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_PyUnicode_TypeRecords' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.8.a(unicodectype.o): In function `gettyperecord':
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/unicodectype.c:55:(.text+0x646c): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.8.a(unicodectype.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_PyUnicode_TypeRecords' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/jn2gb-01/Documents/e/Python-3.8.0/Objects/unicodectype.c:55:(.text+0x64bc): dangerous relocation: unsupported relocation
[... cut ...]
CMakeFiles/Makefile2:1243: recipe for target 'utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/all' failed
make[1]: *** [utils/python/bindings_python_3.8/CMakeFiles/jetson-utils-python-38.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
```

It seems like an error from libpython3.8 (I guess?), but im 100% sure, I install python3.8.0 properly (from source). Well previously this once occurred, then I tried to reinstall python3.8.0, then theres error `cannot find -lnpymath`, now after I resolved the `lnpymath` error, the libpython3.8 error occurred again 😅.

Aldhanekaa, 2023-03-04:
> ah I see. The error is resolved now by installing numpy, thanks for that. However new error occured;
>
> ```
> /usr/bin/ld: /usr/local/lib/libpython3.8.a(abstract.o): relocation [...]

well ChatGPT said, it caused by the mismatch of Python3.8 with my jetson architecture, well after I checked, it matches.

Aldhanekaa, 2023-03-04:
well, just info, I overwrited my  Jetson nano python3.6 with python3.8, fortunately so far there's no system error

Aldhanekaa, 2023-03-04:
I think I found the solution. To overcome this situation, I need to remake the project first (that intended for 3.6 & 3.7, so I delete 3.8 on CMakeLists.txt). Once I remade the project, I type `sudo ldconfig` . Then I tried to add 3.8 in CMakeLists.txt, however theres different new error `-ljetson-utils-python-38`, it already mentioned in this issue https://github.com/dusty-nv/jetson-inference/issues/936 (but still I havent figured out how to overcome the `-ljetson-utils-python-38` error)

Aldhanekaa, 2023-03-04:
oh my bad, 

> I think I found the solution. To overcome this situation, I need to remake the project first (that intended for 3.6 & 3.7, so I delete 3.8 on CMakeLists.txt). Once I remade the project, I type `sudo [...]

that's not the solution, turns out `-ljetson-utils-python-38` error occured because I didnt add 3.8 into CMakeLists.txt inside `utils/python` folder

Aldhanekaa, 2023-03-08:
Ok I think this occurred because Python3.8 is built from source, previously when I try to install jetson inference with python3.7 (built from source) this error occurred however when I try to install jetson inference with python3.7 that was downloaded with apt-get there was no any error

closed 2023-03-08 by Aldhanekaa

jiefengsun, 2023-03-30:
Hello @Aldhanekaa , did you solve all the problems?

jiefengsun, 2023-03-30:
> @jiefengsun yeah I did, what I did was using python3.7 that was installed with apt-get. Well previously I faced the same issue as this with python3.7, then I rebuilt the microsdcsrd image then I install python3.7 with apt-get

Thanks @Aldhanekaa , But I feel the package conflicts is a big issue if 3.7 is installed with apt-get. For example, if I install a numpy, use pip, will it install for python 3.6 or python 3.7. How did you manage this problem?  

Also I found out that using conda virtual environment is not a good idea, because the packages in the Jetpack (eg., TensorRT, cuDNN) cannot be used in the virtual environment.

Aldhanekaa, 2023-03-31:
> Thanks @Aldhanekaa , But I feel the package conflicts is a big issue if 3.7 is installed with apt-get. For example, if I install a numpy, use pip, will it install for python 3.6 or python 3.7. How [...]

ah I see, first thing first if you install package with `pip` it will install the package for the python2.7. So make sure use pip3 for python3

However if you installed python3.7 on Jetson Nano with apt-get, the default python3 version still on 3.6 (if I'm not mistaken, `python3 --version`) , which means anything that you install with pip3 is for python3.6. It is so not recommended to change the default python3 version of Jetson board, so to install library for your python3.7 you need to install it directly `/usr/bin/python3.7 -m pip`

`/usr/bin/python3.7 -m pip install numpy`

Aldhanekaa, 2023-03-31:
> ah I see, first thing first if you install package with `pip` it will install the package for the python2.7. So make sure use pip3 for python3
>
> However if you installed python3.7 on Jetson Nano with [...]

For your information, so far I couldn't install Jetson Inference on python3.7 virtualenv, what I did was let Jetson Inference library installed globally then I

`virtualenv env -p /usr/bin/python3.7 --system-site-packages`

Aldhanekaa, 2023-03-31:
> @jiefengsun yeah I did, what I did was using python3.7 that was installed with apt-get. Well previously I faced the same issue as this with python3.7, then I rebuilt the microsdcsrd image then I install python3.7 with apt-get

Well by the way, when I encountered this issue `Cannot find -lnpymath when building for Python 3.8` , sure it is necessary to uninstall the python3.8, however it's so difficult to uninstall because at the time I installed Python3.8 by building it from source, it is not easy as uninstall with apt-get. So since I was struggling to uninstall python3.8, I decided to rebuilt the image of my Jetson board 😅 . Installing python3 version by building from source gives me a trauma 😅😅