dusty-nv/jetson-inference #1574: What's the correct way to uninstall jetson-inference

jet-c-21, 2023-03-04:
I follow this way to build jetson-inference:
```
sudo apt-get update
sudo apt-get install git cmake libpython3-dev python3-numpy
git clone --recursive https://github.com/dusty-nv/jetson-inference
cd jetson-inference
mkdir build
cd build
cmake ../
make -j$(nproc)
sudo make install
sudo ldconfig
```
And how can I completely uninstall it?

dusty-nv, 2023-03-04:
@jet-c-21 you have to `sudo rm` the relevant files from:

* `/usr/local/bin`
* `/usr/local/include`
* `/usr/lib/python3.8/dist-packages`  (or `python3.6` if you're on JetPack 4)

closed 2023-04-27 by dusty-nv

Bear528, 2024-03-01:
> @jet-c-21 you have to `sudo rm` the relevant files from:
> 
> * `/usr/local/bin`
> * `/usr/local/include`
> * `/usr/lib/python3.8/dist-packages`  (or `python3.6` if you're on JetPack 4)

relevant files is? or delect all files?