dusty-nv/jetson-inference #1583: cmake error

Troyechild, 2023-03-22:
Hello, this is the second time that I have deployed jetson inference, but I have some problems this time.
--Configuring incomplete,errors occurred!
See also "/home/m217/jetson-inference/build/CMakeFiles/CMakeOutput.log"
I don't know why this mistake happened, please help me[](url)
[CMakeOutput.log](https://github.com/dusty-nv/jetson-inference/files/11039709/CMakeOutput.log)

dusty-nv, 2023-03-22:
Hi @Troyechild, can you copy/paste the console terminal log from when you run `cmake ../` instead of CMakeOutput.log?   I can't actually tell the error(s) from CMakeOutput.log

Also, for the time being perhaps you could run the jetson-inference docker container which has it pre-compiled.

Troyechild, 2023-04-03:
> Hi @Troyechild, can you copy/paste the console terminal log from when you run `cmake ../` instead of CMakeOutput.log? I can't actually tell the error(s) from CMakeOutput.log
> 
> Also, for the time being perhaps you could run the jetson-inference docker container which has it pre-compiled.
I solved this problem, and the reason for the problem was that cuda and other libraries were not completely downloaded

Thank you for your prompt reply

Troyechild, 2023-04-03:
> Hi @Troyechild, can you copy/paste the console terminal log from when you run `cmake ../` instead of CMakeOutput.log? I can't actually tell the error(s) from CMakeOutput.log
> 
> Also, for the time being perhaps you could run the jetson-inference docker container which has it pre-compiled.

I found that I had not finished downloading cuda and other libraries. Now I have solved these problems. Thank you for your timely reply

closed 2023-04-27 by dusty-nv