dusty-nv/jetson-inference #1587: how can I simultaneously run object detection and semantic segmentation with SegNet (street detection (fcn-resnet18-cityscapes)) through kammera stream

hosseinAT, 2023-03-24:
I have a question, I have trained a new objects (wheelchair and wheelchair user) and the car should be able to recognize the wheelchair and wheelchair user when it enters the road.

In order to meet the criteria, one must develop a system that recognizes Wheelchair, Wheelchair_User and distinguishes it from other people. Here are some steps to achieve that:

1. Image acquisition systems: Positioning the camera on the car roof, looking in the direction of travel, to monitor the road and capture images of people, Wheelchair_User and Wheelchair.
2. Object Detection Algorithms: Using machine learning or computer vision algorithms to detect wheelchairs and people in the images.
3. Street detection (fcn-resnet18-cityscapes): by which one can know whether the wheelchair and the wheelchair user are on the street or sidewalk (footpath).

how can I simultaneously run object detection and semantic segmentation with SegNet (street detection (fcn-resnet18-cityscapes)) through kammera stream? when i want to run both, i get an error saying i should stop one of them. can you please help me?

https://github.com/hosseinAT/Rollstuhl_und_Rollstuhlfahrererkennung_verwendung_auf_Kriterien.git

dusty-nv, 2023-03-24:
Hi @hosseinAT, you should make your own Python or C++ program that uses both `detectNet` and `segNet` on the same camera image.   I would pass `overlay=none` to detectNet.Detect() so that it doesn't overwrite the original image before processing it with segnet.  Then after both the inference is complete, use detectNet.Overlay()

You can also see this example which does multiple DNNs - https://github.com/dusty-nv/jetson-inference/blob/master/docs/webrtc-flask.md

hosseinAT, 2023-03-28:
Hello Dustin,

Thanks for the support and I tried for a few days but unfortunately it
didn't work.

this is my training model:
 Rollstuhl_und_Rollstuhlfahrererkennung_verwendu...
<https://drive.google.com/file/d/1bHtQTiVNdGoO9WG1zLIDBNTHqFE_EdLu/view?usp=drive_web>

Best regards

hossein