dusty-nv/jetson-inference #1592: Running example programs on docker and installation miniconda on Jetson

hlmhlr, 2023-03-30:
Hi, 

I have the Jetson Xavier NX development kit. After completing the installation of Jetson with JetPack, I chose to use the "Running the Docker Container" option to start with DL/ML model inferences. After executing the following commands, I am trying to run example programs. For this, I am typing `cd build/aarch64/bin` but there is no `build` folder inside `jetson-inference` folder.

```
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ docker/run.sh
```
What am I missing, and how can I resolve this? Besides `Running the Docker Container` and `Building the Project from Source` options, would it be recommendable to install MiniConda on Jetson and then create the virtual environments to run the various repositories? Which option is better to go with for the installations so that various repositories can be run without any conflict of versions, etc.?

I shall be thankful for any prompt and positive response!
Kind regards,

dusty-nv, 2023-03-30:
Hi @hlmhlr, the jetson-inference examples/apps are already installed in the container under `/usr/local/bin` so you should just be able to run things like `video-viewer --help` or `detectnet.py --help` without needing to navigate to `/jetson-inference/build/aarch64/bin`

Otherwise though - do you have a `/jetson-inference` dir in the container and a `/jetson-inference/build` dir in the container?

hlmhlr, 2023-03-31:
Hi @dusty-nv, many thanks for your prompt response.

I am able to locate the `jetson-inference/build` dir now. Also, I have finished with the example programs. Many thanks!! Also, any suggestions for the second part of the question, please?

dusty-nv, 2023-03-31:
I haven't used miniconda/virtualenv on Jetson and recommend the container approach instead

hlmhlr, 2023-03-31:
Thanks @dusty-nv for your response!!

Since I am a newbie with Dockers, I hope my basic queries will not bother you!
 
I am facing an issue while running `Coding Your Own Image Recognition Program (Python)` using the `Docker container`. 

1. I tried to mount the device using the command below:

```hlr@hlr-desktop:~/jetson-inference$ docker/run.sh -it -v /home/hlr/my-recognition-py:/home/hlr/jetson-inferences/my-recognition-py```
The output of this execution is shown as:
 
[image]

How do i know that the folder from the host has been mounted or not? I tried this [link](https://stackoverflow.com/questions/39620404/docker-check-within-the-container-if-a-directory-is-mounted-from-the-host-or-n)  by running the command from the docker container 
```mount | grep '/home/hlr/jetson-inferences/my-recognition-python'```
 
and it throws the output as:
```/dev/nvme0n1p1 on /home/hlr/jetson-inferences/my-recognition-python type ext4 (rw,relatime)```

I am unable to interpret this regarding the successful mounting of the folder or not? Please help me in clarifying this. 

2. To run the program mentioned [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-python-2.md),  `$ ./my-recognition.py my_image.jpg` command needs to be executed from within the container or outside?
In both cases, how to run `sudo make install` command as it is throwing the message `make: *** No rule to make target 'install'.  Stop.`. 

Your input would be valuable to me.

dusty-nv, 2023-03-31:
It looks like it was successfully mounting it, it that `/home/hlr/jetson-inferences/my-recognition-py` shows up inside your container at all, then it was successful.    You should run the program from inside the container.  Also, I don't think you need to do `make` since this is a python script.

hlmhlr, 2023-04-06:
Yup, it worked now. Many thanks again @dusty-nv!

closed 2023-04-06 by hlmhlr