dusty-nv/jetson-inference #1593: PeopleNet for live detection with camera

jpgarbiso, 2023-03-30:
Hello, I'm trying to use detectnet for a live pedestrian detection demo with a camera. However, even though in detectnet-console-2.md TAO PeopleNet is listed as an available pre-trained detection model, it cannot be downloaded (not listed) in the Model Downloader tool, nor is it in the mirror file list.
That is why I've been looking for other options, like what we see in https://forums.developer.nvidia.com/t/tao-and-jetson-inference-ooops/240989 --> After seeing this, I tried downloading the PeopleNet model and installing the TAO Converter. I managed to convert the PeopleNet model to an .engine file, but at the end when I run the example command provided by dusty-nv, calling detectnet with the .engine file and the associated labels for the --model and --labels arguments respectively, the other arguments unchanged, and /dev/video0 at the end instead of the images route, I get:

[TRT] invalid engine cache file size (0 bytes)
[TRT] failed to load
[TRT] detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network
Traceback (most recent call last):
File "./detectnet.py", line 55, in <module>
net = detectNet(args.network, sys.argv, args.threshold)
Exception: jetson.inference -- detectNet failed to load network

Does someone have an idea of what I'm doing wrong? Or at least how I could get to download the PeopleNet pre-trained model with the downloader tool?

dusty-nv, 2023-03-30:
If you are using a recent build of jetson-inference, there is no longer the Model Downloader tool needed - the models will be downloaded automatically by jetson-inference library when they are run.  Including the TAO models like peoplenet.  Sorry, i need to update the docs about this.

dusty-nv, 2023-03-30:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-tao.md

mister-owl, 2023-03-30:
Hi @dusty-nv ! I had tried using peoplenet directly but it didn't recognize the model name. I flashed this Nano's SD card recently though, and installed the latest versions of everything as far as I know. Anyway, I see you replied via e-mail so you probably didn't see the update I added in the original post. I managed to use the converted model (PeopleNet, pruned), but it runs at 7 FPS only (compared to 20+ when using MobileNet SSD v2 at the same resolution) and it looks pretty bad at detecting people. While MobileNet SSD v2 easily detected people in front of the camera, PeopleNet struggles to do so. I thought this might be because PeopleNet might be trained for a view from afar rather than staight in front of the camera, so I tried putting the camera outside the window and it does not detect any pedestrian at all when they pass. I'm on a sixth floor though (I guess it's too high), but I'm using the webcam's native resolution (1280x720). Do you think this performance is to be expected? (significant drop in FPS, much worse detection performance facing the camera, no detection at all from a 6th floor at 720p)

dusty-nv, 2023-03-30:
@mister-owl on Nano I believe the FPS drop is expected because it runs in FP16 and the PeopleNet resolution is much higher (~900x750 vs 300x300).   On Xavier/Orin, it runs in INT8 and has equal or better performance.

Regarding the detection performance, in my experience the PeopleNet is much better....can you try lowering the confidence threshold?

mister-owl, 2023-03-30:
@dusty-nv I reduced the threshold to .3 but anyway I discovered that it had a terribly poor performance before because the camera was tilted and I was too close. Aligning the camera so that people are at 90° with respect to the bottom of the frame instead of ~60° improved the detection performance a lot. And unlike MobileNetSSDv2, which is able to detect a person pretty close to the camera, PeopleNet seems to be able to detect only with full-body view, or almost-full-body view. With only half of the body, it's undetectable even with a .3 threshold.

A probably silly question (since I'm a newbie in this). I downoladed the PeopleNet model that was labeled as INT8, and I think I stated INT8 in every place that I had to specify during the process. It is still passed to FP16 when used in the Nano? And another thing. Regardless of the input resolution from the webcam, performance will necessarily be tied to the image size that the model uses originally? (i.e. 900x750)

dusty-nv, 2023-03-30:
Yes, it will use FP16 on Nano (even though the filename says INT8), and yes the input resolution to the model will be ~900x750 regardless of what your camera resolution is

closed 2023-04-27 by dusty-nv

creativesh, 2023-12-21:
Hello
If Icwant to use peoplenet model on ubuntu 20, (not jetson) , what should I do ? I have tested some ways and they have not resulted

dusty-nv, 2023-12-21:
> If Icwant to use peoplenet model on ubuntu 20, (not jetson) , what should I do ? I have tested some ways and they have not resulted

https://github.com/dusty-nv/jetson-inference/issues/1767#issuecomment-1866476392