dusty-nv/jetson-inference #1596: Running detectnet on mac and/or PC for testing and training?

joeysantana3, 2023-04-06:
I'm trying to train a network for my specific task using my own dataset.I'm following the [guide here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md). Currently, I'm performing training on a combination of different machines depending on where I am. Either Google Colab, a desktop with Ubuntu 20.04 and a 3080Ti, or a Mac.

The problem is, every time I want to test the trained network, I have to transfer it to the Jetson Nano, convert it to onnx format, then run detectnet and let the model "compile" (not sure thats what its called). This takes a lot of time just to test the performance and see the model's output. Ultimately I will run the model on the nano, but I need to get it trained to an acceptable point first.

**Is there a way I can run the SSD model locally using only python just to test out its performance?**

I say only python because I can't figure out how to compile the jetson library on other machines and I think detectnet.py just uses the c++ version somehow. I tried digging through the code to see if I could figure it out myself. I was able to load the model checkpoint and pass an image through, but the output is not what I'm expecting and I don't know how to get the output to the format I want (bounding boxes to input in to cv2.rectangle).

Here is a snippet of code that I copied and modified a bit
```python
class_names = [name.strip() for name in open(labels_file).readlines()]
num_classes = len(class_names)
device = torch.device('cuda:0' if torch.cuda.is_available() else 'cpu')
net = create_mobilenetv1_ssd(len(class_names), is_test=False)
net.load(model_file)
net.to(device)
predictor = create_mobilenetv1_ssd_predictor(net, candidate_size=200)

root_path = '/model/JPEGImages'

for image_path in os.listdir('/model/JPEGImages'):
    orig_image = cv2.imread(os.path.join(root_path, image_path))
    image = cv2.cvtColor(orig_image, cv2.COLOR_BGR2RGB)
    boxes, labels, probs = predictor.predict(image, 10, 0.01)

    # debug print statements
    print(f"boxes: {boxes}")
    print(f"labels: {labels}")
    print(f"probs: {probs}")
    for i in range(boxes.size(0)):
        box = boxes[i, :]
        cv2.rectangle(orig_image, (int(box[0]), int(box[1])), (int(box[2]), int(box[3])), (255, 255, 0), 4)

    cv2.imshow("image", orig_image)
    cv2.waitKey(0)
    cv2.destroyAllWindows()

```

Here is an example of the output of the print statements:
```
boxes: tensor([[ 1079.1694,  2483.1111, -7681.9834, -6872.5283]])
labels: tensor([1])
probs: tensor([0.4852])
```

The above code "works", meaning I can pass an image through it and get results without exceptions. But as I mentioned, the outputs are not in a format I know how to apply to a video/image.

Here is the full python file that the above code was taken from. I essentially just copied everything from other parts of the repo into one file to see if I could get it to work. It's messy, but might be helpful: https://gist.github.com/joeysantana3/9fe1fe9535915d13724f837ea1b6c281

I suppose all I really need is a way to convert what I'm already getting into bounding box locations, but I'm sure there is a better way to run a custom trained ssd model just to see it's output. Whatever info you can provide is greatly appreciated. Thank you!

dusty-nv, 2023-04-06:
Hi @joeysantana3, you can try the `run_ssd_example.py` or `eval_ssd.py` scripts to test inference from the PyTorch checkpoints:

* https://github.com/dusty-nv/pytorch-ssd/blob/86155c0c410e0959df0184b24af6a8f59f49fbe5/run_ssd_example.py
* https://github.com/dusty-nv/pytorch-ssd/blob/86155c0c410e0959df0184b24af6a8f59f49fbe5/eval_ssd.py

Also if you run train_ssd.py with the `--validate-mean-ap` flag, it will print out overall and per-class accuracies after each training epoch.

If you have Linux x86_64 and the NVIDIA Container Runtime installed, you can run jetson-inference container on x86 with the same `docker/run.sh` script.   However using detectnet/detectnet.py will still have TensorRT "compile" the ONNX model.  

Hope that helps!

joeysantana3, 2023-04-07:
Hi @dusty-nv, I appreciate you taking the time to respond. I know this repo is pretty active.

I can't believe I didn't check the `run_ssd_example.py` script out before posting this. It seems this script will do what I need. But, I ran in to another issue that I wanted to let you know about. Take a look at script args I ran and the output:
Command:
`python run_ssd_example.py mb1-ssd ~/Downloads/desktopmodel/model/mb1-ssd-Epoch-299-Loss-2.360474967956543.pth ~/Downloads/desktopmodel/model/labels.txt ~/Downloads/datasets/project41103/JPEGImages/2023-04-01-115428_frame_1000_scale_0.3.jpg`

Output:
```
Traceback (most recent call last):
  File "/Users/joey/source/jetson-inference/python/training/detection/ssd/run_ssd_example.py", line 55, in <module>
    cv2.rectangle(orig_image, (box[0], box[1]), (box[2], box[3]), (255, 255, 0), 4)
cv2.error: OpenCV(4.5.4) :-1: error: (-5:Bad argument) in function 'rectangle'
> Overload resolution failed:
>  - Can't parse 'pt1'. Sequence item with index 0 has a wrong type
>  - Can't parse 'pt1'. Sequence item with index 0 has a wrong type
>  - Can't parse 'rec'. Expected sequence length 4, got 2
>  - Can't parse 'rec'. Expected sequence length 4, got 2
```

I made sure the ssd repo (which I think is a submodule) was clean by doing 
`git reset --hard 86155c0c410e0959df0184b24af6a8f59f49fbe5`

I added in a print statement:
`print(f"(box[0]: {box[0]}, box[1]: {box[1]}, box[2]: {box[2]}, box[3]: {box[3]})")`

And got this output:
`(box[0]: 671.8101196289062, box[1]: 322.60308837890625, box[2]: 816.9866333007812, box[3]: 400.1157531738281)`

I think cv2.rectangle only takes ints, so I modified the line to be this:
`cv2.rectangle(orig_image, (int(box[0]), int(box[1])), (int(box[2]), int(box[3])), (255, 255, 0), 4)`

It seems to be working now! I'm going to do a little more testing, and then I'll close this issue. Again, thanks for responding, I'm sure everyone running into issues appreciates it!

dusty-nv, 2023-04-07:
OK, thanks @joeysantana3, glad you got it working!   `run_ssd_example.py` was a script that came from the upstream fork of pytorch-ssd, and I haven't maintained it - I think it was originally authored against an older version of OpenCV.

closed 2023-04-27 by dusty-nv