dusty-nv/jetson-inference #1597: is it possible to directly load a torch tensor?

Yiming0416, 2023-04-07:
Hi,

Thanks for your great work! I am wondering if there is a way to directly load the input image from torch tensor? 

I saw in `jeston-utils` you can convert cuda image to pytorch tensor. Is there a way to convert pytorch tensor to cuda image?

dusty-nv, 2023-04-08:
I haven't entirely been able to figure that out yet but have some ideas based on torch.tensor.data_ptr() , or torch's implementation of __cuda_array_interface__

For the time being, I would continue using the numpy/ndarray route.

dusty-nv, 2023-04-27:
@Yiming0416 support for this has been added in https://github.com/dusty-nv/jetson-utils/commit/c692b89f5716e01823ff177e286eba6986bc4c6a

closed 2023-04-27 by dusty-nv