dusty-nv/jetson-inference #1598: pull video stream from rtsp issue

UserName-wang, 2023-04-08:
same rtsp address, detectnet works fine, but detectnet.py cannot work.
#./detectnet rtsp://admin:pass]*@192.168.0.33:554/stream1

this will not work:
#./detectnet.py rtsp://admin:pass]*@192.168.0.33:554/stream1 
detectnet.py rtsp://admin:pass]*@192.168.0.33:554/stream1 
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for admin:pass]*@192.168.0.33
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- resource discovery and auto-negotiation failed
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:Lner]N71*@192.168.0.33:554/stream1
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.15) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 51, in <module>
    input = videoSource(args.input, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device

UserName-wang, 2023-04-08:
import jetson_inference
import jetson_utils
from jetson_utils import videoSource, videoOutput, Log

camera = jetson_utils.videoSource("rtsp://admin:pass]*@192.168.0.33:554/stream1")      # '/dev/video0' for V4L2 or "csi://0" 
display = jetson_utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
	img = camera.Capture()
	display.Render(img)

error message:
/usr/lib/python3.8/dist-packages/jetson_inference/__init__.py
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstDecoder -- creating decoder for admin:pass]*@192.168.0.33
[gstreamer] gstDecoder -- failed to discover stream info
[gstreamer] gstDecoder -- resource discovery and auto-negotiation failed
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:pass]*@192.168.0.33:554/stream1
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.15) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "/home/nx/study/NVIDIA/jetson-inference/my_projects/test.py", line 5, in <module>
    camera = jetson_utils.videoSource("rtsp://admin:pass]*@192.168.0.33:554/stream1")      # '/dev/video0' for V4L2 or "csi://0" 
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2023-04-08:
Are there extra ]* characters in your URL?   Can you try video-viewer.py ?

camera = ***@***.***:554/stream1") # '/dev/video0' for V4L2 or "csi://0"

[gstreamer] gstDecoder -- failed to create decoder for ***@***.***:554/stream1
camera = ***@***.***:554/stream1") # '/dev/video0' for V4L2 or "csi://0"

UserName-wang, 2023-04-09:
@dusty-nv ， really thank you for your quick reply!
seems this issue were caused by the remote rtsp camera server. the connection is unstable.
I can connect to the rtsp via both detectnet and detectnet.py sometimes. and seems the disconnection function in the scripts not closed properly. sometimes I  have to turn off the camera to get the connection to rtsp. Do you have suggestion If I want to use OpenCV and Gstreamer ? thank you again!